PX4/PX4-Avoidance #354: Quesntion on ThreePointMsg.msg support

Jaeyoung-Lim, 2019-05-14:
`ThreePointMsg.msg` used in global_planner seems to be doing something quite similar to what `mavros_msgs/trajectory` is sending.

Would it make sense to depricate `ThreePointMsg` in global planner? or is there a reason that this is needed?

mhkabir, 2019-05-14:
What was the conclusion here?

mrivi, 2019-05-15:
@Jaeyoung-Lim  why did you close the issue? 
`ThreePointMsg` is old  message that was never upstreamed. It can be removed

Jaeyoung-Lim, 2019-05-15:
@mrivi @mhkabir I was confused on what it was actually doing, so closed it so that I wanted to better understand what it is doing.

So, I need some help here, `ThreePointMsg` seems to be used to communicate between the `global_planner_node` and `path_handler_node`. Do you mean that this interface is not being used anymore?

mrivi, 2019-05-15:
@Jaeyoung-Lim  I am going by memory here but if I remember correctly it's published but not used anywhere.

Jaeyoung-Lim, 2019-05-15:
@mrivi Okay, that's a good hint. I will look more into it. I guess it makes sense to deprecate it if it is not used.

PR #362 (2019-05-16): Deprecate threePointMsg in global_planner
This PR deprecates the threePointMsg custom message implementation in `global_planner`.

Initially it was quite confusing for me to understand why this message exists in the first place. It seems like it was a interface that was a precursor of the Avoidance triplet. This Fixes #354 

This does not affect the performance of global_planner.

Tested in SITL, both in offboard and mission
closed 2019-05-16 by jkflying