PX4/PX4-Avoidance #361: General issue for global_planner suggestion

acpl00, 2019-05-16:
Dear people

I have tested the global_planner with the command 
`$roslaunch global planner_sitl_mavros.launch` in the outdoor environment the drone is suppose to work and I had some issues:

-  The drone suddenly stop during flying  the planned path.
-  When the drone got into the goal position after executing the planned path, it suddenly disarmed, then I armed it again  and it kept arm for a short period   and then it disarmed again.

Well, I guess if these issues happen in real simulation, the drone will crash.

- Just a general question :)
  - What shall be the best way to use global_planner algorithm in a real simulation?.  I mean, Is  the best way to use it together with `QgroundControlStation`? Or,  just `RVIZ` and the real drone.

  -  What is the safest way to use` global_planner` algorithm?

 - I guess, I need a secure mechanism in case global planner fail, right? 

Thank you

mrivi, 2019-05-16:
@acpl00 please do not post duplicate issues.

closed 2019-05-16 by mrivi

acpl00, 2019-05-16:
I am sorry I forgot to delete the other one, but thank you for the replay :)