PX4/PX4-Avoidance #363: docker build local_planner, global_planner failed

TianmingLi, 2019-05-16:
**Describe the bug**

./run.py local --build
./run.py global --build

failed

**To Reproduce**
Steps to reproduce the behavior:
1. clone and checkout last avoidance repository
2. follow deployment instructions of docker https://github.com/PX4/avoidance/tree/master/docker#deploying-with-docker
3. ./run.py local --build failed,  ./run.py local --run failed also
4. ./run.py global --build failed

**Expected behavior**
build local error is 

```
07059dab833c
Step 10/13 : RUN ["/bin/bash", "-c", "    source /opt/ros/kinetic/setup.bash &&     catkin build -w ${CATKIN_WS} --cmake-args -DCMAKE_BUILD_TYPE=Release "]
 ---> Running in 50eac9d25d84
NOTICE: Could not determine the width of the terminal. A default width of 80 will be used. This warning will only be printed once.
-------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/kinetic
Workspace:                   /root/catkin_ws
-------------------------------------------------------
Build Space:        [exists] /root/catkin_ws/build
Devel Space:        [exists] /root/catkin_ws/devel
Install Space:      [unused] /root/catkin_ws/install
Log Space:         [missing] /root/catkin_ws/logs
Source Space:       [exists] /root/catkin_ws/src
DESTDIR:            [unused] None
-------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-------------------------------------------------------
[build] Found '1' packages in 0.0 seconds.                                     
[build] Updating package table.                                                
Starting >>> catkin_tools_prebuild                                             
Finished <<< catkin_tools_prebuild                [ 2.3 seconds ]              
Starting >>> local_planner                                                     
_______________________________________________________________________________
Errors << local_planner:cmake /root/catkin_ws/logs/local_planner/build.cmake.000.log
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "mavros" with any
  of the following names:

    mavrosConfig.cmake
    mavros-config.cmake

  Add the installation prefix of "mavros" to CMAKE_PREFIX_PATH or set
  "mavros_DIR" to a directory containing one of the above files.  If "mavros"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  CMakeLists.txt:10 (find_package)

CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "mavros" with any
  of the following names:

    mavrosConfig.cmake
    mavros-config.cmake

  Add the installation prefix of "mavros" to CMAKE_PREFIX_PATH or set
  "mavros_DIR" to a directory containing one of the above files.  If "mavros"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  CMakeLists.txt:10 (find_package)

cd /root/catkin_ws/build/local_planner; catkin build --get-env local_planner | catkin env -si  /usr/bin/cmake /root/catkin_ws/src/local-planner --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/root/catkin_ws/devel/.private/local_planner -DCMAKE_INSTALL_PREFIX=/root/catkin_ws/install -DCMAKE_BUILD_TYPE=Release; cd -
...............................................................................
Failed << local_planner:cmake                   [ Exited with code 1 ]         
Failed <<< local_planner                        [ 2.3 seconds ] 

```

build global error is 

```
Errors << global_planner:cmake /root/catkin_ws/logs/global_planner/build.cmake.000.log
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "mavlink" with any
  of the following names:

    mavlinkConfig.cmake
    mavlink-config.cmake

  Add the installation prefix of "mavlink" to CMAKE_PREFIX_PATH or set
  "mavlink_DIR" to a directory containing one of the above files.  If
  "mavlink" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:9 (find_package)

CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "mavlink" with any
  of the following names:

    mavlinkConfig.cmake
    mavlink-config.cmake

  Add the installation prefix of "mavlink" to CMAKE_PREFIX_PATH or set
  "mavlink_DIR" to a directory containing one of the above files.  If
  "mavlink" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:9 (find_package)

cd /root/catkin_ws/build/global_planner; catkin build --get-env global_planner | catkin env -si  /usr/bin/cmake /root/catkin_ws/src/global-planner --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/root/catkin_ws/devel/.private/global_planner -DCMAKE_INSTALL_PREFIX=/root/catkin_ws/install -DCMAKE_BUILD_TYPE=Release; cd -
...............................................................................
Failed << global_planner:cmake                  [ Exited with code 1 ]  
```

mrivi, 2019-05-17:
Hi, thank you for reporting this. The avoidance package is undergoing a lot of development and we aren't able to keep the containers available. Therefore we are going to deprecate them.
However if you are interested in running the avoidance in Docker you can use this container https://github.com/PX4/containers/blob/master/docker/px4-dev/Dockerfile_ros-kinetic .

TianmingLi, 2019-05-24:
px4io/px4-dev-ros-kinetic container works, 
but it's also not out-of-the-box and run, 
need to  add `mavros`  to catkin workspace to make the build pass
https://discuss.px4.io/t/local-planner-build-error/8874

mrivi, 2019-05-24:
@TianmingLi you don't need mavros in the workspace. It's installed from the package manager https://github.com/PX4/containers/blob/master/docker/px4-dev/Dockerfile_ros-kinetic#L28

TianmingLi, 2019-05-26:
thanks for the reply，

maybe something goes wrong，

i try two computer to get `px4io/px4-dev-ros-kinetic container`  and use apt to check the `mavros` version in container, one is 0.29, another is 0.26, but all didn't have `CompanionProcessStatus.h` file, i double check it through `ls /opt/ros/kinetic/include/` and `find / -name "CompanionProcessStatus.h"`, did not find the header file, it's weird

mrivi, 2019-05-26:
what version of px4io/px4-dev-ros-kinetic are you running? Any tag after `px4io/px4-dev-ros-kinetic:2019-03-08` should be fine since we are running the CI on it. 
What we do to run the avoidance inside the container is in this script https://github.com/PX4/Firmware/blob/master/test/rostest_avoidance_run.sh

The script is fine for the avoidance tag 0.2.0.. If you want to use avoidance master you need to adapt the system variables since the avoidance has changed structure since the 0.2.0 release

closed 2019-05-29 by TianmingLi