PX4/PX4-Avoidance #376: Local planner. Limit angular velocity

urock, 2019-05-23:
Hi!
I am working indoor autonomous navigation. Using Realsense T265 as positioning source, RealSense D435 for obstacle detection  and px4/avoidance/local_planner for safe navigation on predetermined waypoints. 

While testing I noticed that to move around obstacles drone is turning too fast so that T265 SLAM might loose its position causing navigation failure. 

Is there a way to limit angular velocity in local_planner? Right now I am inspecting the code for the best place to implement it. 

Any suggestions would be welcomed! Thanks!

jkflying, 2019-05-23:
Hi! There is already some code to smooth the angular velocity, but nothing limiting it. You need to look in the Waypoint Generator https://github.com/PX4/avoidance/blob/b23b8bf2a46effd0750880dfcd73566fcc9c9cf2/local_planner/src/nodes/waypoint_generator.cpp#L214, I guess it wouldn't be too difficult to add this.

Also, it would be great to get the parameter for what the limit should be from the flight controller - this way we can set the parameter in QGC.

urock, 2019-05-23:
Julian, thanks for the hint! I see that reducing 
`smoothing_speed_xy_` should help me. 
What Flight controller parameter to check? I might be able to implement it and make PR

jkflying, 2019-05-23:
`smoothing_speed_xy_` will also make your translation smoother, so be careful, it might mean it is so smooth it can't get around obstacles in time.

You need the `MC_YAWRATE_MAX` parameter. And thanks, a PR would be great.

jkflying, 2019-07-03:
This can also be done in the flight controller, setting the max yaw rate.

closed 2019-07-03 by jkflying