PX4/PX4-Avoidance #383: Launch Error: environment variable 'GAZEBO_RESOURCE_PATH' is not set

Jaeyoung-Lim, 2019-05-29:
**Describe the bug**
After #381, The following error appears when launching a launchfile
```
~/catkin_ws/src/avoidance$ roslaunch global_planner global_planner_stereo.launch
... logging to /home/jaeyoung/.ros/log/298a61c6-8202-11e9-bcc9-002315d18521/roslaunch-jaeyoung-ThinkPad-T480-30070.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

environment variable 'GAZEBO_RESOURCE_PATH' is not set
The traceback for the exception was written to the log file
```

**To Reproduce**
Steps to reproduce the behavior:
1. Delete `GAZEBO_RESOURCE_PATH` or `GAZEBO_MODEL_PATH` from environment
2. Launch `global_planner_stereo.launch`

**Expected behavior**
Models should be able to be imported from the correct path

@nicovanduijn When you said you had a problem, have you tried with deleting the `GAZEBO_MODEL_PATH` from your environmnet?

nicovanduijn, 2019-05-29:
I have, yes. It worked for me, which was the reason for #381 

Did it work for you prior to #381 and then no longer?

Jaeyoung-Lim, 2019-05-29:
Yes, It worked for me prior to #381 and it is not working anymore. Reverting #381 works for me.

Wierd!

mrivi, 2019-05-31:
In the PX4 #issue channel they had the same issue 
```bash
environment variable 'GAZEBO_RESOURCE_PATH' is not set
```

Can we revert this one? Think more about this change... Setting the `GAZEBO_MODEL_PATH` and `GAZEBO_RESOURCE_PATH` doesn't ensure the planners running correctly since the ` ~/Firmware/Tools/setup_gazebo.bash` also sets them.

nicovanduijn, 2019-05-31:
I really don't understand what's going on. It seems that what works on my machine is orthogonal to what works on Jay's...

nicovanduijn, 2019-05-31:
It looks like my machine is the odd one, and pre- #381 works for most. I'll revert it and figure out what's wrong on my machine

mrivi, 2019-07-01:
fixed by https://github.com/PX4/avoidance/pull/392

closed 2019-07-01 by mrivi