PX4/PX4-Avoidance #389: Collision Prevention without local planner

hansbonini, 2019-05-30:
Hi guys, i'm trying to configure de collision prevention over the obstacle avoidance module in PX4.
I've remapped my guidance obstacle distance topic to "/mavros/obstacle/send" and setted the parameter MPC_COL_PREV_D on the FCU.

There are more topics to publish? Or only these one ("/mavros/obstacle/send") is needed?
Can anybody give me a path? What i forget?

mrivi, 2019-05-31:
You only need to send distance data trough that topic and you are good to go. Then start your rosnode, switch into position control, control the vehicle with a gamepad.

hansbonini, 2019-05-31:
@mrivi i will try here. thank you so much for clear my doubts.

closed 2019-06-03 by mrivi

bhaskar-glitch, 2023-08-20:
> @mrivi i will try here. thank you so much for clear my doubts.

Hi @hansbonini Did u tried the same on drone?