PX4/PX4-Avoidance #393: failed to launch global planner

TianmingLi, 2019-05-31:
avoidance version is `a442ce8b91dc853b35b7f9e4c1328b40d2fe2eeb`
px4 firmware version is ` v1.9.0`

 fallow the instruction

in one terminal start avoidance through roslaunch

```
cd ~/catkin_ws/src
source ~/catkin_ws/devel/setup.bash
. ~/Firmware/Tools/setup_gazebo.bash ~/Firmware ~/Firmware/build/px4_sitl_default

export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:~/Firmware

roslaunch global_planner global_planner_stereo.launch
```

another terminal to start px4 simulator
```
make px4_sitl_default gazebo
```

if i start avoidance first and then start px4 simulator, i got error in gazebo/server/default.log
```
Gazebo multi-robot simulator, version 7.0.0
Copyright (C) 2012-2016 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

(1559300344 186449642) [Msg] Waiting for master.
(1559300344 188320391) [Msg] Connected to gazebo master @ http://127.0.0.1:11345
(1559300344 188499088) [Msg] Publicized address: 10.0.2.15
(1559300347 451804930) Init world[vrc_task_1]
(1559300347 508106501) [Wrn] [ModelDatabase.cc:339] Getting models from[http://gazebosim.org/models/]. This may take a few seconds.
(1559300407 664840709) [Err] [ModelDatabase.cc:413] Unable to download model[model://iris_stereo_camera]
(1559300407 665383840) [Err] [SystemPaths.cc:410] File or path does not exist[""]
(1559300407 771376271) [Err] [World.cc:1911] Unable to find a model, light, or actor in:

```

vice versa，start px4 simulator first, then start avoidance,  got error

in gazebo/server/default.log
```
Gazebo multi-robot simulator, version 7.0.0
Copyright (C) 2012-2016 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

(1559300544 440732699) [Msg] Waiting for master.
(1559300544 454086309) [Msg] Connected to gazebo master @ http://127.0.0.1:11345
(1559300544 455001584) [Msg] Publicized address: 10.0.2.15
(1559300545 176234864) Init world[default]
(1559300545 207658140) [Wrn] [gazebo_barometer_plugin.cpp:66] [gazebo_barometer_plugin] Using default home altitude of 488 m
(1559300545 217258622) [Dbg] [gazebo_mavlink_interface.cpp:135] <joint_name> not found for channel[0] no joint control will be performed for this channel.
(1559300545 217838400) [Dbg] [gazebo_mavlink_interface.cpp:135] <joint_name> not found for channel[1] no joint control will be performed for this channel.
(1559300545 218207444) [Dbg] [gazebo_mavlink_interface.cpp:135] <joint_name> not found for channel[2] no joint control will be performed for this channel.
(1559300545 219926986) [Dbg] [gazebo_mavlink_interface.cpp:135] <joint_name> not found for channel[3] no joint control will be performed for this channel.
(1559300545 220480951) [Wrn] [gazebo_mavlink_interface.cpp:124] joint [zephyr_delta_wing::propeller_joint] not found for channel[4] no joint control for this channel.
(1559300545 221687508) [Wrn] [gazebo_mavlink_interface.cpp:124] joint [zephyr_delta_wing::flap_left_joint] not found for channel[5] no joint control for this channel.
(1559300545 222247379) [Wrn] [gazebo_mavlink_interface.cpp:124] joint [zephyr_delta_wing::flap_right_joint] not found for channel[6] no joint control for this channel.
(1559300545 222841184) [Dbg] [gazebo_mavlink_interface.cpp:135] <joint_name> not found for channel[7] no joint control will be performed for this channel.
(1559300545 223311563) [Msg] Connecting to PX4 SITL using TCP
(1559300545 223596522) [Msg] Lockstep is enabled
(1559300545 224051599) [Msg] Speed factor set to: 1
(1559300547 523137551) [Msg] Using MAVLink protocol v2.0
(1559300547 949401058) [Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/iris/motor_speed/0, deleting message. This warning is printed only once.
(1559300548 957929459) [Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/iris/motors, deleting message. This warning is printed only once.
(1559300548 958466374) [Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/iris/motor_speed/1, deleting message. This warning is printed only once.
(1559300548 958914086) [Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/iris/motor_speed/2, deleting message. This warning is printed only once.
(1559300548 959317969) [Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/default/iris/motor_speed/3, deleting message. This warning is printed only once.

```

in roslaunch terminal 
```
process[gazebo-7]: started with pid [21907]
[sitl_0-5] process has died [pid 21905, exit code 255, cmd /home/li/Firmware/build/px4_sitl_default/bin/px4 /home/li/Firmware/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -i 0 -d __name:=sitl_0 __log:=/home/li/.ros/log/a229af5a-8393-11e9-9c39-080027917f9c/sitl_0-5.log].
log file: /home/li/.ros/log/a229af5a-8393-11e9-9c39-080027917f9c/sitl_0-5*.log

[gazebo-7] process has died [pid 21907, exit code 255, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode /home/li/catkin_ws/src/avoidance/avoidance/sim/worlds/simple_obstacle.world __name:=gazebo __log:=/home/li/.ros/log/a229af5a-8393-11e9-9c39-080027917f9c/gazebo-7.log].
log file: /home/li/.ros/log/a229af5a-8393-11e9-9c39-080027917f9c/gazebo-7*.log

```

mrivi, 2019-05-31:
`roslaunch global_planner global_planner_stereo.launch` already starts PX4 sitl. That is the only command required to use the simulation.

TianmingLi, 2019-05-31:
https://github.com/PX4/avoidance/blob/5aa633d1a16bd9c662ec1c5b890caf4122d54a31/global_planner/launch/global_planner_stereo.launch#L19
i checked above launch file and thought `roslaunch global_planner global_planner_stereo.launch` will start px4 and gazebo, but actually it didn't happen, gazebo didn't showup and px4 sitl also not show. 
only get a rviz loaded

and in `~/.gazebo/server/default.log`
got same error 

```
Gazebo multi-robot simulator, version 7.0.0

(1559300407 664840709) [Err] [ModelDatabase.cc:413] Unable to download model[model://iris_stereo_camera]

```

but is the `model://iris_stereo_camera` in the folder of `avoidance\sim\models\iris_stereo_camera\model.config` ? i dont know why gazebo not find it

TianmingLi, 2019-05-31:
i copy all models from  `avoidance\sim\models\`  to `~/.gazebo/models` 
then run `roslaunch global_planner global_planner_stereo.launch`

the `(1559300407 664840709) [Err] [ModelDatabase.cc:413] Unable to download model[model://iris_stereo_camera]` disappear

get some new error

```
Gazebo multi-robot simulator, version 7.0.0
Copyright (C) 2012-2016 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

(1559316531 502004745) [Msg] Waiting for master.
(1559316531 510142325) [Msg] Connected to gazebo master @ http://127.0.0.1:11345
(1559316531 510292455) [Msg] Publicized address: 10.0.2.15
(1559316534 355539342) Init world[vrc_task_1]
(1559316535 314671189) [Wrn] [msgs.cc:1655] Conversion of sensor type[multicamera] not suppported.
(1559316536 511468870) applying Gaussian noise model with mean 0, stddev 0.007, bias -0
(1559316536 521022589) applying Gaussian noise model with mean 0, stddev 0.007, bias -0
(1559316536 935080136) [Wrn] [gazebo_barometer_plugin.cpp:66] [gazebo_barometer_plugin] Using default home altitude of 488 m
(1559316536 970393083) [Dbg] [gazebo_mavlink_interface.cpp:135] <joint_name> not found for channel[0] no joint control will be performed for this channel.
(1559316536 970553412) [Dbg] [gazebo_mavlink_interface.cpp:135] <joint_name> not found for channel[1] no joint control will be performed for this channel.
(1559316536 970617911) [Dbg] [gazebo_mavlink_interface.cpp:135] <joint_name> not found for channel[2] no joint control will be performed for this channel.
(1559316536 970699323) [Dbg] [gazebo_mavlink_interface.cpp:135] <joint_name> not found for channel[3] no joint control will be performed for this channel.
(1559316536 970812608) [Wrn] [gazebo_mavlink_interface.cpp:124] joint [zephyr_delta_wing::propeller_joint] not found for channel[4] no joint control for this channel.
(1559316536 970916443) [Wrn] [gazebo_mavlink_interface.cpp:124] joint [zephyr_delta_wing::flap_left_joint] not found for channel[5] no joint control for this channel.
(1559316536 971094720) [Wrn] [gazebo_mavlink_interface.cpp:124] joint [zephyr_delta_wing::flap_right_joint] not found for channel[6] no joint control for this channel.
(1559316536 971192390) [Dbg] [gazebo_mavlink_interface.cpp:135] <joint_name> not found for channel[7] no joint control will be performed for this channel.
(1559316536 971249540) [Msg] Connecting to PX4 SITL using TCP
(1559316536 971277897) [Msg] Lockstep is enabled
(1559316536 971304079) [Msg] Speed factor set to: 1
(1559316537 896259958) [Msg] Using MAVLink protocol v2.0
(1559316538 76606787) [Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/vrc_task_1/iris/motors, deleting message. This warning is printed only once.
(1559316538 76767592) [Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/vrc_task_1/iris/motor_speed/0, deleting message. This warning is printed only once.
(1559316538 76828533) [Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/vrc_task_1/iris/motor_speed/1, deleting message. This warning is printed only once.
(1559316538 76888295) [Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/vrc_task_1/iris/motor_speed/2, deleting message. This warning is printed only once.
(1559316538 76946962) [Wrn] [Publisher.cc:140] Queue limit reached for topic /gazebo/vrc_task_1/iris/motor_speed/3, deleting message. This warning is printed only once.
(1559316538 148701030) [Wrn] [msgs.cc:1655] Conversion of sensor type[multicamera] not suppported.
(1559316539 656160173) The link iris::iris::base_link has unrealistic inertia, unable to visualize box of equivalent inertia.
```
also  gazebo die

```
[gazebo-7] process has died [pid 9778, exit code 255, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode /home/li/catkin_ws/src/avoidance/avoidance/sim/worlds/simple_obstacle.world __name:=gazebo __log:=/home/li/.ros/log/561f5c82-83ba-11e9-8b9e-080027917f9c/gazebo-7.log].

```

nicovanduijn, 2019-05-31:
Try running
```
echo export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:~/catkin_ws/src/avoidance/avoidance/sim/models:~/catkin_ws/src/avoidance/avoidance/sim/worlds >> ~/.bashrc
```

and source your bashrc: 
```
source ~/.bashrc
```

mrivi, 2019-05-31:
The planner doesn't start the gazebo gui so you will only see rviz when the planner is working properly. 
Most likely your system variables are not properly set up. Try what @nicovanduijn suggested.

TianmingLi, 2019-06-01:
for simplfiy, i just copy `models` and `worlds` folder to `~/.gazebo/`
and re-launch, everything seems goes fine, rviz load, tree and drone show, until gazebo die
```
gzserver: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreRenderSystem.cpp:546: virtual void Ogre::RenderSystem::setDepthBufferFor(Ogre::RenderTarget*): Assertion `bAttached && "A new DepthBuffer for a RenderTarget was created, but after creation" "it says it's incompatible with that RT"' failed.
[ INFO] [1559349159.460470833, 2.624000000]: FCU: Avoidance system connected
Aborted (core dumped)
[gazebo-7] process has died [pid 3175, exit code 134, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode /home/li/catkin_ws/src/avoidance/avoidance/sim/worlds/simple_obstacle.world __name:=gazebo __log:=/home/li/.ros/log/a6239d66-8404-11e9-adb2-080027917f9c/gazebo-7.log].
```

it seems the reason of virtualbox, so, i will try it in real machine

TianmingLi, 2019-06-01:
in the physical machine, after solve two more error

```
ERROR: cannot launch node of type [rqt_reconfigure/rqt_reconfigure]: rqt_reconfigure
# sudo apt-get install ros-kinetic-rqt-reconfigure

ERROR: cannot launch node of type [stereo_image_proc/stereo_image_proc]: stereo_image_proc
# sudo apt install ros-kinetic-stereo-image-proc
```

everything goes well 

thank your guys

closed 2019-06-01 by mrivi