PX4/PX4-Avoidance #397: Custom avoidance algorithm in PX4 firmware (as a new module)

alexcherpi, 2019-06-04:
In order to be more real-time and not to embed a companion computer, would it be possible to have a custom collision avoidance algorithm in PX4 firmware as a PX4 module (thread)?
Moreover, instead of being in 'offboard' would it be possible to be in 'manual' or in 'position' (velocity) mode in order to add the avoidance commands to the input commands of the user (simultaneously) ?

mrivi, 2019-06-04:
@alexcherpi  There is already Collision Prevention in Position Control Mode in the PX4 Firmware. See https://docs.px4.io/en/computer_vision/collision_prevention.html

At the moment, you would need the companion to get the distance data. However we are working on making the collision prevention work with distance sensor directly connected to the FCU. It should be ready fairly soon.

alexcherpi, 2019-06-04:
Hi @mrivi, thanks, I already read the information about the collision_prevention. 
Regarding the connection of the distance sensor directly to the FCU, what is the link of this PR ?

mrivi, 2019-06-04:
Collision prevention runs only in position control. You need the companion to get the distance data. 

Please note that [this PR](https://github.com/PX4/Firmware/pull/10998) isn't finished nor tested

closed 2019-06-07 by mrivi