PX4/PX4-Avoidance #403: Local_planer no fly in mission mode

MKargus0, 2019-06-10:
When I am try to start mission on my quad with local_planer it ignore the mission (Mission mode is active but drone not fly by trajectory)
It Can be controlled by goals and local planer work
I see that topic mavros/trajectory /desired    is empty
COM_OBS_AVOID is set up to 1 
Without obstacle avoidance mission flying well
What do you think about this problem? Thank you.

mrivi, 2019-06-11:
@MKargus0, can you please share the Flight Control log? You can upload it here https://review.px4.io/ and share the link. Thank you

MKargus0, 2019-06-11:
You can see this one ,if it will be little , I can send new
https://review.px4.io/plot_app?log=aee600ed-8c5d-4eda-8d1d-c41af8366236

[image]

mrivi, 2019-06-17:
The flight controller is continuously getting into Loiter becuase it doesn't receive messages from the companion. Please check that mavlink and mavros are properly setup.
Do you have also the console output of the local planner?

alexisnov, 2019-06-20:
Thank you, @mrivi! 
What type of messages do you mean? 
We have full connection between companion and PX4 FCU. 
We are able to make commands over Mavlink from companion computer, we are able to set goals and our quad flyes with avoidence,  we get all data from PX4,  except trajectory.
Can you explain more how is it possible?

MKargus0, 2019-06-21:
@mrivi  ,Thank you.
This is console output of local_planer

[consol output.txt](https://github.com/PX4/avoidance/files/3314736/consol.output.txt)

and flight log
https://review.px4.io/plot_app?log=e62d13e8-9f7d-4474-b8dd-d705eeac7482

mrivi, 2019-06-22:
Can you enable the SDLOG_PROFILE with the avoidance data from QGC when you fly? 
Is the avoidance actually running? I see a lot of Realsense errors in the console output you have shared.

gajena, 2019-06-25:
@MKargus0 To receive `/mavros/trajectory/desired` you have to arm the vehicle.

MKargus0, 2019-06-27:
I have solved this problem. Pixhawk was conected to companion computer via  USB and when we try to fly in mission mode  ,px4 dont send trajectory messeges. Now companion computer and pixhawk conected by UART ,telem1 port on pixawk . In this configuration we recive trajectory messeges. Thank you for your help.
But I don t understend why usb conection don t allow to send this messeges. Can you tell me why? Thanks

jkflying, 2019-07-09:
I'm not sure why, that sound like some configuration in the Pixhawk.

closed 2019-07-09 by jkflying