PX4/PX4-Avoidance #407: Land and RTL issue while avoidance is active

RyanHurst, 2019-06-12:
**Describe the bug**
Land and Return modes don't seem to behave any differently than hold mode while avoidance is enabled.

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on
2. Performed mission
3. Shifted to RTL
4. Drone didn't move

**Expected behavior**
I would expect the drone to start looking around for a safe way back to the take off location.

**Log Files and Screenshots**
https://review.px4.io/plot_app?log=0b6f155d-e95f-4a7c-84a0-fb9a55694776

**Additional context**
After finishing my misison I told the drone to RTL. The drone shifted into RTL, but didnt' move. I shifted back to HOLD, then RTL again. Drone didn't move. I then shifted into land. The drone didn't move. Then I set COM_OBSTACLE_AVOID and then told the drone to land and it landed.

mrivi, 2019-06-13:
Hi, thanks for reporting. The issue with auto land is know, we're going to fix it as soon as possible. RTL was working until a while ago. We need to look into it.

mrivi, 2019-07-10:
possible fix: PR #445

mrivi, 2019-08-14:
Fixex in #445

closed 2019-08-14 by mrivi