PX4/PX4-Avoidance #412: unable to build global_planner

ramb182k, 2019-06-18:
I have installed octomap using
$sudo apt install ros-kinetic-octomap
still when i tried to build, building global_planner has this error

$ catkin build                                    
------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/kinetic
Workspace:                   /home/ramb182k/ws
------------------------------------------------------
Build Space:        [exists] /home/ramb182k/ws/build
Devel Space:        [exists] /home/ramb182k/ws/devel
Install Space:      [unused] /home/ramb182k/ws/install
Log Space:          [exists] /home/ramb182k/ws/logs
Source Space:       [exists] /home/ramb182k/ws/src
DESTDIR:            [unused] None
------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------
[build] Found '4' packages in 0.0 seconds.                                     
[build] Package table is up to date.                                           
Starting  >>> avoidance                                                        
Finished  <<< avoidance                           [ 21.0 seconds ]             
Starting  >>> global_planner                                                   
Starting  >>> local_planner                                                    
Starting  >>> safe_landing_planner                                             
_______________________________________________________________________________
Errors     << global_planner:make /home/ramb182k/ws/logs/global_planner/build.make.004.log
In file included from /home/ramb182k/ws/src/avoidance/global_planner/src/library/global_planner.cpp:1:0:
/home/ramb182k/ws/src/avoidance/global_planner/include/global_planner/global_planner.h:19:34: fatal error: octomap_msgs/Octomap.h: No such file or directory
compilation terminated.
make[2]: *** [CMakeFiles/global_planner.dir/src/library/global_planner.cpp.o] Error 1
make[1]: *** [CMakeFiles/global_planner.dir/all] Error 2
make: *** [all] Error 2
cd /home/ramb182k/ws/build/global_planner; catkin build --get-env global_planner | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << global_planner:make                 [ Exited with code 2 ]       
Failed    <<< global_planner                      [ 35.2 seconds ]             
Finished  <<< safe_landing_planner                [ 1 minute and 22.6 seconds ]
Finished  <<< local_planner                       [ 1 minute and 22.4 seconds ]
[build] Summary: 3 of 4 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 1 minute and 44.3 seconds total. 

what should i do now?

Jaeyoung-Lim, 2019-06-18:
As written in the Readme, you can install the octomap_msgs using the following.
```
sudo apt install ros-kinetic-octomap-msgs
```

ramb182k, 2019-06-18:
thanks.it works! :)

closed 2019-06-18 by ramb182k