PX4/PX4-Avoidance #415: RealSense D430 module supported? (non-RGB D435)

NicksonYap, 2019-06-20:
Hi!

I just found out about the RealSense D430 module (with the same USB-C connection as D435)

It seems that D430 is the same module used inside D435, with additional RGB camera.

(not my pic)
[image]

[image]

[image]

It's has a much smaller form factor and can be separated from the main sensor circuitry
It's also 30-40% cheaper...

Does the repo depend on RGB input, depth only or both?
such that we can use the non-RGB D430 instead
What would be needed to be configured in the software?

Best Regards

Nickson
Founder
[seamlesshq.com](url)

mrivi, 2019-06-20:
@NicksonYap - All planners require a pointCloud as input. We don't have any other requirement

NicksonYap, 2019-06-20:
@mrivi
Ah yes
To be specific, they are xyz point clouds
non-rgb right?

mhkabir, 2019-06-20:
@NicksonYap yes.

closed 2019-06-21 by mrivi

kafan1986, 2019-10-25:
@NicksonYap Where did you get the D400 interposer ribbon cable from?

bsprime, 2021-02-07:
> Hi!
>
> I just found out about the RealSense D430 module (with the same USB-C connection as D435)
>
> It seems that D430 is the same module used inside D435, with additional RGB camera.
>
> (not my pic) [...]

@NicksonYap where did you get interposer ribbon cable.