PX4/PX4-Avoidance #416: Avoidance system is not working with different system IDs.

gajena, 2019-06-24:
I am testing the avoidance system for collision avoidance between multi MAVs while carrying out a collaborative mission. So first I started with single MAV and avoidance system worked fine. Then, I started testing with two MAVs :
* First MAV (who has system ID 1) worked well as expected.
* Second MAV (system ID 2, I have also set ) not arming at all, and keep on giving a warning:- Avoidance system is not ready.
* Then tested only with single MAV but with system ID 2 and it was not working.

# Parameters

```shell
SUMMARY
========

CLEAR PARAMETERS
 * /uav0/pilot/
 * /uav1/pilot/

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /uav0/pilot/cmd/use_comp_id_system_control: False
 * /uav0/pilot/conn/heartbeat_rate: 1.0
 * /uav0/pilot/conn/system_time_rate: 1.0
 * /uav0/pilot/conn/timeout: 10.0
 * /uav0/pilot/conn/timesync_rate: 10.0
 * /uav0/pilot/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /uav0/pilot/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /uav0/pilot/distance_sensor/hrlv_ez4_pub/id: 0
 * /uav0/pilot/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /uav0/pilot/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /uav0/pilot/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /uav0/pilot/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /uav0/pilot/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /uav0/pilot/distance_sensor/laser_1_sub/id: 3
 * /uav0/pilot/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /uav0/pilot/distance_sensor/laser_1_sub/subscriber: True
 * /uav0/pilot/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /uav0/pilot/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /uav0/pilot/distance_sensor/lidarlite_pub/id: 1
 * /uav0/pilot/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /uav0/pilot/distance_sensor/lidarlite_pub/send_tf: True
 * /uav0/pilot/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /uav0/pilot/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /uav0/pilot/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /uav0/pilot/distance_sensor/sonar_1_sub/id: 2
 * /uav0/pilot/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /uav0/pilot/distance_sensor/sonar_1_sub/subscriber: True
 * /uav0/pilot/fake_gps/eph: 2.0
 * /uav0/pilot/fake_gps/epv: 2.0
 * /uav0/pilot/fake_gps/fix_type: 3
 * /uav0/pilot/fake_gps/geo_origin/alt: 408.0
 * /uav0/pilot/fake_gps/geo_origin/lat: 47.3667
 * /uav0/pilot/fake_gps/geo_origin/lon: 8.55
 * /uav0/pilot/fake_gps/gps_rate: 5.0
 * /uav0/pilot/fake_gps/mocap_transform: True
 * /uav0/pilot/fake_gps/satellites_visible: 5
 * /uav0/pilot/fake_gps/tf/child_frame_id: fix
 * /uav0/pilot/fake_gps/tf/frame_id: map
 * /uav0/pilot/fake_gps/tf/listen: False
 * /uav0/pilot/fake_gps/tf/rate_limit: 10.0
 * /uav0/pilot/fake_gps/tf/send: False
 * /uav0/pilot/fake_gps/use_mocap: True
 * /uav0/pilot/fake_gps/use_vision: False
 * /uav0/pilot/fcu_protocol: v2.0
 * /uav0/pilot/fcu_url: udp://:14540@loca...
 * /uav0/pilot/gcs_url: 
 * /uav0/pilot/global_position/child_frame_id: base_link
 * /uav0/pilot/global_position/frame_id: map
 * /uav0/pilot/global_position/gps_uere: 1.0
 * /uav0/pilot/global_position/rot_covariance: 99999.0
 * /uav0/pilot/global_position/tf/child_frame_id: base_link
 * /uav0/pilot/global_position/tf/frame_id: map
[... 222 lines cut ...]
 * /uav1/pilot/vision_pose/tf/frame_id: map
 * /uav1/pilot/vision_pose/tf/listen: False
 * /uav1/pilot/vision_pose/tf/rate_limit: 10.0
 * /uav1/pilot/vision_speed/listen_twist: True
 * /uav1/pilot/vision_speed/twist_cov: True
 * /uav1/pilot/wheel_odometry/child_frame_id: base_link
 * /uav1/pilot/wheel_odometry/count: 2
 * /uav1/pilot/wheel_odometry/frame_id: map
 * /uav1/pilot/wheel_odometry/send_raw: True
 * /uav1/pilot/wheel_odometry/send_twist: False
 * /uav1/pilot/wheel_odometry/tf/child_frame_id: base_link
 * /uav1/pilot/wheel_odometry/tf/frame_id: map
 * /uav1/pilot/wheel_odometry/tf/send: True
 * /uav1/pilot/wheel_odometry/use_rpm: False
 * /uav1/pilot/wheel_odometry/vel_error: 0.1
 * /uav1/pilot/wheel_odometry/wheel0/radius: 0.05
 * /uav1/pilot/wheel_odometry/wheel0/x: 0.0
 * /uav1/pilot/wheel_odometry/wheel0/y: -0.15
 * /uav1/pilot/wheel_odometry/wheel1/radius: 0.05
 * /uav1/pilot/wheel_odometry/wheel1/x: 0.0
 * /uav1/pilot/wheel_odometry/wheel1/y: 0.15
 * /uav1/rotors_description: <?xml version="1....
 * /use_sim_time: True

```

dayjaby, 2019-06-24:
I experienced the same issue without px4/avoidance whilst trying the multi uav launch file from PX4/Firmware. Setting the target_system to 1 for all drones made it work. Can you try that?

Jaeyoung-Lim, 2019-06-24:
Could you provide more information on how you are launching multiple instances of the avoidance?

1. which planner are you using? the global planner or the local planner?
2. Could you provide the launchfile you were using or explain how you were launching multiple instances of avoidance? In the log provided I don't see any avoidance node starting

nicovanduijn, 2019-06-24:
I'm like 90% sure this is related to the MAVROS heartbeat

gajena, 2019-06-24:
@Jaeyoung-Lim I am running my custom planner for avoidance in mission mode (available [here](https://github.com/gajena/imav_avoidance/blob/master/avoidance/src/avoidance_node.cpp#L1))

Whats going on in custom planner
* Subscribed the other MAVs global coordinate, velocities and attitude in a custom msg.
* Then calculate the time to get hit by other MAV and if `time to hit` will go beyond the time threshold, MAVs receives modified plans. 

* subscribed from mavros
  * /mavros/trajectory/desire

* published to mavros
  * /mavros/trajectory/generated
  * /mavros/companion_process/status

## launch for MultiMAV
```xml
<?xml version="1.0"?>
<launch>
    <!-- MAVROS posix SITL environment launch script -->
    <!-- launches Gazebo environment and 2x: MAVROS, PX4 SITL, and spawns vehicle -->
    <!-- vehicle model and world -->
    <arg name="est" default="ekf2"/>
    <arg name="vehicle" default="iris"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/empty.world"/>
    <!-- gazebo configs -->
    <arg name="gui" default="true"/>
    <arg name="debug" default="false"/>
    <arg name="verbose" default="false"/>
    <arg name="paused" default="false"/>
    <!-- Gazebo sim -->
    <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="gui" value="$(arg gui)"/>
        <arg name="world_name" value="$(arg world)"/>
        <arg name="debug" value="$(arg debug)"/>
        <arg name="verbose" value="$(arg verbose)"/>
        <arg name="paused" value="$(arg paused)"/>
    </include>
    <!-- UAV0 -->
    <group ns="uav0">
        <!-- MAVROS and vehicle configs -->
        <arg name="ID" value="0"/>
        <arg name="fcu_url" default="udp://:14540@localhost:14580"/>
        <!-- PX4 SITL and vehicle spawn -->
        <include file="$(find px4)/launch/single_vehicle_spawn.launch">
            <arg name="x" value="0"/>
            <arg name="y" value="0"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="mavlink_udp_port" value="14560"/>
            <arg name="mavlink_tcp_port" value="4560"/>
            <arg name="ID" value="$(arg ID)"/>
        </include>
        <!-- MAVROS -->
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value=""/>
            <arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>
            <arg name="tgt_component" value="1"/>
        </include>
    </group>
    <!-- UAV1 -->
    <group ns="uav1">
        <!-- MAVROS and vehicle configs -->
        <arg name="ID" value="1"/>
        <arg name="fcu_url" default="udp://:14541@localhost:14581"/>
        <!-- PX4 SITL and vehicle spawn -->
        <include file="$(find px4)/launch/single_vehicle_spawn.launch">
            <arg name="x" value="1"/>
            <arg name="y" value="0"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="mavlink_udp_port" value="14561"/>
            <arg name="mavlink_tcp_port" value="4561"/>
            <arg name="ID" value="$(arg ID)"/>
        </include>
        <!-- MAVROS -->
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value=""/>
            <arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>
            <arg name="tgt_component" value="1"/>
        </include>
    </group>

</launch>
```

## luanch for custom avoidance
```xml
<?xml version="1.0"?>
<launch>
    <group ns="uav0">
    <node name="avoidance_node" type="avoidance_node" pkg="avoidance" output="screen">
        <param name="mis_obs" value="false"/>
        <param name="t_to_hit_thresh" value="5"/>
        <param name="mav_id" value="1"/>
    </node>
    </group>
    <group ns="uav1">
    <node name="avoidance_node" type="avoidance_node" pkg="avoidance" output="screen">
        <param name="mis_obs" value="false"/>
        <param name="t_to_hit_thresh" value="5"/>
        <param name="mav_id" value="2"/>
    </node>
    </group>
</launch>
```

dayjaby, 2019-06-24:
```<arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>```

Can you change the value here to "1" for both and report if it works?

gajena, 2019-06-24:
@dayjaby 

> `<arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>`
> 
> Can you change the value here to "1" for both and report if it works?

I tried this, and it did not work.

## Probable issue
* the Second iris starts with MAV_ID 2 in firmware param and by setting it 1 in mavros will result in lost communication.

Jaeyoung-Lim, 2019-06-24:
@gajena I still don't understand how this is related to this repo. I apologize if I am being completely stupid

gajena, 2019-06-24:
@Jaeyoung-Lim For multiple vehicles setup in mavros and QGC, they require different system IDs for a different system.
So I went like this and face the problem that, avoidance is not working when system ID is changed from default to a different system ID.
So probably looks like Firmware issue, I wanted to know that if someone has tried the multi-vehicle with avoidance.

Jaeyoung-Lim, 2019-06-24:
@gajena local_planner has been tested with different sysIDs. Have you tried running the local_planner?

gajena, 2019-06-25:
@Jaeyoung-Lim I checked and everything works in offboard mode without changing `COM_OBS_AVOID` with different sysIDs. But in mission mode (after setting `COM_OBS_AVOID  = 1`) is not working with different MAV sysIDs. 
And QGC starts to give out warning msg ` avoidance system not available`.
I think it is a Firmware issue so that I will put it there.
Thank you for your support.

closed 2019-06-25 by gajena

Cavalletta98, 2023-01-02:
The planner is still not working with multiple system ID. I'm running with PX4 version 1.13.2 and the latest commit of the avoidance software