PX4/PX4-Avoidance #419: local_planner realsense avoidance system rejected error

ramb182k, 2019-06-25:
this is what i get on my terminal
$ roslaunch local_planner local_planner_realsense.launch
... logging to /home/ramb182k/.ros/log/d7543b3e-975f-11e9-a597-2016b9c6ddc4/roslaunch-ramb182k-G3-3579-11567.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ramb182k-G3-3579:44985/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /local_planner_node/goal_x_param: 17
 * /local_planner_node/goal_y_param: 15
 * /local_planner_node/goal_z_param: 3
 * /local_planner_node/pointcloud_topics: ['/realsense/came...
 * /local_planner_node/world_name: /home/ramb182k/ca...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: local_origin
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: True
 * /mavros/local_position/tf/send_fcu: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: False
 * /mavros/mocap/use_tf: True
[... 585 lines cut ...]
[ WARN] [1561477375.398225631, 19.268000000]:  Planner abort: missing required data 
 
[ WARN] [1561477375.498312837, 19.368000000]:  Planner abort: missing required data 
 
[ WARN] [1561477375.598622323, 19.468000000]:  Planner abort: missing required data 
 
[ WARN] [1561477375.698398141, 19.568000000]:  Planner abort: missing required data 
 
[ WARN] [1561477375.798271798, 19.668000000]:  Planner abort: missing required data 
 
[ WARN] [1561477375.898765239, 19.768000000]:  Planner abort: missing required data 
 
[ WARN] [1561477375.998021294, 19.868000000]:  Planner abort: missing required data 
 
[ WARN] [1561477376.098418820, 19.968000000]:  Planner abort: missing required data 
 
[ WARN] [1561477376.203005597, 20.072000000]:  Planner abort: missing required data 
 
[ WARN] [1561477376.298861534, 20.168000000]:  Planner abort: missing required data 
 
[ WARN] [1561477376.398900642, 20.268000000]:  Planner abort: missing required data 
 
[ WARN] [1561477376.498652705, 20.368000000]:  Planner abort: missing required data 
 
Can you help me with what is wrong here?

mrivi, 2019-06-26:
It seems that the planner never received a pointcloud on the topic ` /realsense/camera/depth/points` .
Is the [nodelet](https://github.com/PX4/avoidance/blob/master/local_planner/launch/local_planner_realsense.launch#L43) running?

ramb182k, 2019-06-26:
yes it is running.....from /standalone_nodelet to /local_planner_node

mrivi, 2019-06-26:
is `/realsense/camera/depth/points` published?

ramb182k, 2019-06-26:
[image]
Yes it is published

Jaeyoung-Lim, 2019-06-26:
@ramb182k `rqt_graph` only shows the subscription / publish relationships. It doesn't mean that any data is going through the topics. Can you check if there is any data within the `/realsense/camera/depth/points` ?

ramb182k, 2019-06-26:
no there is no data 
[image]

mrivi, 2019-06-26:
Have you tried any other launch file (such that we know that you're setup is working properly)?
Could you please also check that the pointcloud isn't on a different topic? Can you also try to visualize the depth image and see if there is data there?

ramb182k, 2019-06-27:
Yes, the depth-camera works fine but there are two warnings.
[image]
 Here the pointcloud is in the topic /local_pointcloud 
[image]

The depth image appears.so no problems

nicovanduijn, 2019-06-27:
If there is a `/local_pointcloud' it means your planner is running, since it's being published in the main thread. At the same time you are saying there is no data on `/realsense/camera/depth/points', which is essentially the input to the planner. This doesn't make sense to me, there cannot be an output point cloud without any sensor data flowing in

ramb182k, 2019-06-28:
No, this pointcloud above was from depth-camera and not realsense.
In realsense the local_pointcloud does't have any data

mrivi, 2019-06-28:
Most likely there is some setup error on the realsense nodlet since the other launch file is running fine

closed 2019-07-09 by jkflying