PX4/PX4-Avoidance #425: Global planner node-9 process died

JoeyLeeNPU, 2019-06-28:
When I use the global planner, I often meet a problem as following:

****
[global_planner_node-9] process has died [pid 5793, exit code -11, cmd /home/nb/avoidance_ws/devel/lib/global_planner/global_planner_node __name:=global_planner_node __log:=/home/nb/.ros/log/42cfa022-9983-11e9-ac6e-0242b966eb14/global_planner_node-9.log].
log file: /home/nb/.ros/log/42cfa022-9983-11e9-ac6e-0242b966eb14/global_planner_node-9*.log
****

The launch file is default, and the environment is Gazebo simulation.
Some parameters we set as "robot radius = 0.5", "resolution = 1.0", "cell scale = 1.0".
And another question is how should I set above parameters, and could tell me the meaning of "cell scale"? 

Waiting for response, thanks.

mrivi, 2019-06-28:
@Jaeyoung-Lim it's a know issue. We are trying to figure it out. If you have any insights, they're the most welcomed.

JoeyLeeNPU, 2019-06-28:
> @Jaeyoung-Lim it's a know issue. We are trying to figure it out. If you have any insights, they're the most welcomed.

Until now, we have tried to adjust these three parameters, we found that the frequency of "process died" is different(**when we set "robot radius = 0.5", "resolution = 1.0 or 0.5", "cell scale = 1.0", the number of died is smaller than other condition**), and this "died" frequency may also related positively to distance between current position("s" in code) and goal("t" in code).

We think that the problem may happen in **data format** because this global planner algorithm uses octomap and cell to represent discrete space and the drone.
Another is that during the path planning process, there are a mount of octomap grid information has been recorded, will it cause a problem such as memory crash?

Thanks so much for your response!

mrivi, 2019-07-01:
FYI @Jaeyoung-Lim

jkflying, 2019-07-01:
@JoeyLeeNPU Have you been able to catch the segfault in a debugger so we can get a line number? Maybe using GDB or Valgrind? I tried to catch the segfault but it wouldn't happen when I ran in the debugger.

JoeyLeeNPU, 2019-07-01:
I meet this problem when I run the global planner and this problem shows in the terminal. I copy the whole error message in my first description. And I don’t use other debugger. So how should I do next?

| |
李卓一 NPU
|
|
邮箱：15929972316@163.com
npulzy
|

签名由 网易邮箱大师 定制

On 07/01/2019 23:11, Julian Kent wrote:

@JoeyLeeNPU Have you been able to catch the segfault in a debugger so we can get a line number? Maybe using GDB or Valgrind? I tried to catch the segfault but it wouldn't happen when I ran in the debugger.

jkflying, 2019-07-01:
If you install `valgrind` and add `launch-prefix="valgrind"` to the global planner node then it should run in valgrind.
It is better explained here:
http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB

Then if it crashes it will show the exact line number where the crash happened. Note that the execution is much slower though.

JoeyLeeNPU, 2019-07-01:
I will try it now. And I want to konw if you met this error in gazebo when run global planner?

| |
李卓一 NPU
|
|
邮箱：15929972316@163.com
npulzy
|

签名由 网易邮箱大师 定制

On 07/01/2019 23:21, Julian Kent wrote:

If you install valgrind and add launch-prefix="valgrind" to the global planner node then it should run in valgrind.
It is better explained here:
http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB

Then if it crashes it will show the exact line number where the crash happened. Note that the execution is much slower though.

jkflying, 2019-07-01:
Yes, I saw it when using Global Planner with RViz for vizualization and Gazebo headless. However when I tried to use valgrind I couldn't get the problem to reproduce.

JoeyLeeNPU, 2019-07-01:
I see... and when I test this pkg on my drone, it also crashes same as gazebo, the flight will be influenced. Do u have some suggestions? Thanks very very much.

| |
李卓一 NPU
|
|
邮箱：15929972316@163.com
npulzy
|

签名由 网易邮箱大师 定制

On 07/01/2019 23:30, Julian Kent wrote:

Yes, I saw it when using Global Planner with RViz for vizualization and Gazebo headless. However when I tried to use valgrind I couldn't get the problem to reproduce.

jkflying, 2019-07-01:
Another option would be to enable Thread Sanitizer and Address Sanitizer in the compiler. Then when the crash happens hopefully we get a stack trace.

JoeyLeeNPU, 2019-07-01:
Okay, I will try as you said. If I have some thoughts I will reply you quickly. And if you fix this bug, I will update my package in time.
Thanks again.

| |
李卓一 NPU
|
|
邮箱：15929972316@163.com
npulzy
|

签名由 网易邮箱大师 定制

On 07/01/2019 23:37, Julian Kent wrote:

Another option would be to enable Thread Sanitizer and Address Sanitizer in the compiler. Then when the crash happens hopefully we get a stack trace.

JoeyLeeNPU, 2019-07-04:
In these days, I have a try on debug this problem both in gazebo and on my real drone, but sorry to say that I have not solve it.
I guess that the problem comes from the memory management module when the "global_planner_node" is running because the error number is "-11".
If any "while" function or any "callback" function may have a memory problem, such as too many data has pushed into stack waiting to process? 
I think use two threads to process data like "local_plnanner" would be better.

I added the yaw control based on your way-point position control law because there are too many constrains in slam-based flight.

jkflying, 2019-07-10:
This is fixed in #441

closed 2019-07-10 by jkflying

JoeyLeeNPU, 2019-07-10:
Very very thanks to you all! I will update my package and try it on my drone.
And I see there is a "to do" in yaw control, we have written a yaw control law in order to steer slam-based drone. 
The strategy is that if the angle between the reference and the current is too large, I will order the drone turn first until the desired position appears in FOV, then drone flies to the target position with adjusting heading to the reference yaw.  

Thanks to U all again ! hah~