PX4/PX4-Avoidance #427: OFFBOARD Timed out!, No GPS fix, Model: none

optimize0107, 2019-07-03:
Hello.. help me plz.. i'm tired. I reinstalled ubuntu 7 times..

ubuntu 16.04
Gazebo7
ROS Kinetic

1. If 'roslaunch~~' first
    (1) Can't see drone, stereo images in Rviz, Gazebo
    (2) No GPS fix(and i can't use 'commander takeoff')
    (3) Timed out!. when use 'mode -c OFFBOARD'

2. If 'make px4~~' first
    (1) Can see drone in Gazebo, Can activate 'fcu', 'Vehicle pose' in Rviz(but can't see drone)
    (2) Can't see trees
    (3) Cant' synchronize time between Gazebo, Rviz(but i can use 'commander takeoff')
    (3) Timed out!. when use 'mode -c OFFBOARD'

===============================================================
If  run 'roslaunch' first, (I heard that 'make px4~~'is not nessesary)
- [terminal 1]
       
      roslaunch global_planner global_planner_stereo.launch
     
    then some trees show up, but there's no drone. And 'No GPS fix' warning
      [image]

- [terminal 2]
[image]

- [terminal 3]
     I can't see the drone in the gazebo too
[image]

[image]
If use 'commander takeoff' in PX4, 
[image]

- When i ready for this simulation. I used(I downloaded 'Firmware' to ~/catkin_ws/src/Firmware, not ~/Firmware)

      . ~/catkin_ws/src/Firmware/Tools/setup.bash ~/catkin_ws/src/Firmware ~/catkin_ws/src/Firmware/build/px4_sitl_default
   
      export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:~catkin_ws/src/Firmware
     
      echo export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:~/catkin_ws/src/avoidance/avoidance/sim/models:~/catkin_ws/src/avoidance/avoidance/sim/worlds >> ~/.bashrc
      

===============================================================
If  run 'make px4' first,
- [terminal 1]
[image]

- [terminal 2]
[image]
[image]
[image]

I can see 'fcu' and 'Vehicle pose', but i can't see these when i started 'roslaunch' first.
No matter how long i wait, Rviz Time is not connected to Gazebo Time(Radeon RX580, Ryzen 5 2600, 16GB RAM). But strangley, when i use 'commander takeoff' in PX4, both Gazebo and Rviz screens are move up. If i run 'mode -c OFFBOARD', 'Timed out!' show up.

[image]
================================================================

 **I really want to see this!**
[image]

mrivi, 2019-07-03:
@optimize0107 `roslaunch global_planner global_planner_stereo.launch` already launches PX4 sitl. You don't need to run `make px4_sitl gazebo` as well.
You have to run `make px4_sitl gazebo` once to build the Firmware. Then you stop the simulation and run only `roslaunch global_planner global_planner_stereo.launch`

optimize0107, 2019-07-03:
@mrivi Yes i read that in other issue. But running 'make px4_sitl gazebo' was just for checking drone

My problem is that when I run only 'roslaunch global_planner global_planner_stereo.launch ',
1) I can't sea drone in Rviz.
2) No GPS fix  warning.
3) Can't change mode to OFFBOARD by timed out error

mrivi, 2019-07-03:
@optimize0107 - Could you please share the output of `echo` on the following environment variables in the terminal where you're trying to run the global planner
- ROS_PACKAGE_PATH
- GAZEBO_MODEL_PATH
- GAZEBO_RESOURCE_PATH
- GAZEBO_PLUGIN_PATH
- LD_LIBRARY_PATH

optimize0107, 2019-07-04:
@mrivi WOW.  I can simulate global_planner. Thanks!
I don't know exact reason. But I think PATH was the problem. 
I downloaded 'Firmware' at '~/catkin_ws/src/Firmware'. So I deleted them(Files in 'build', 'src' related to 'Firmware') and do same things with '~/Firmware'. 
[image]

closed 2019-07-04 by optimize0107

aydal, 2019-07-21:
Hello @mrivi

siyuada, 2019-07-24:
@aydal Hey, I got the same issue, and fixed by add px4 lib path to ~/.bashrc.
add the line below to your ~/.bashrc, and change the abs path.
`
export LD_LIBRARY_PATH=:/home/uav514/catkin_ws/devel/lib:/opt/ros/kinetic/lib:/opt/ros/kinetic/lib/x86_64-linux-gnu:/home/uav514/Firmware/Tools/setup_gazebo.bash:/home/uav514/Firmware:/home/uav514/Firmware/build/px4_sitl_default
`

Hope it helps.