PX4/PX4-Avoidance #447: Use tags or markers to mark safe area for Safe Landing

jkflying, 2019-07-11:
Currently the planner only looks for flat surfaces to land. It would be good if it could also (or instead) choose a visual target (eg. AprilTag, ChiliTag, [ArUcO](https://docs.opencv.org/master/d5/dae/tutorial_aruco_detection.html#gsc.tab=0) etc) and land on that instead.

For better integration maybe this could use some kind of identifier from the mission item itself, and only land on a tag with the same ID.

khbnzr, 2020-01-23:
@jkflying @mrivi Do we have any progress on the visual target for safe landing ? Please let me know.

jkflying, 2020-01-23:
No progress on this.

khbnzr, 2020-01-24:
@jkflying What was your approach to replace the flat surface with the tags or markers ? I want to implement the tag functionality with the safe landing planner node.

jkflying, 2020-01-24:
It would be roughly like this:

- Start with existing safe landing planner
- Descend until laser range finder gives the desired altitude above ground (instead of point cloud data from depth camera)
- Fly the same grid pattern as the safe landing planner, but at each grid point, instead of looking for a flat area, check if the camera image contains the April/aruco/whatever tag you are looking for.
- Calculate tag location based on laser altitude and angle (from camera focal length)
- Reposition so that the vehicle is directly above the calculated tag location. Repeat until within required accuracy bounds
 - Descend, while keeping the calculated tag location directly below.

If you are going to contribute this upstream I'd be happy to provide additional guidance as you go.

akkawimo, 2021-03-06:
@jkflying What's the status with this? Is it still something that you'd like to add? I could work on it.