PX4/PX4-Avoidance #464: [Avoidance SITL] How to get PX4 ulog

SwiftGust, 2019-08-13:
When performing Avoidance SITL, I cannot find where firmware ulog is located.
In sitl_gazebo, ulog should be located in as in such as {path_to_px4_firmware}/build/px4_sitl_default/logs/ 

However, I couldn't find any ulog generated from Avoidance SITL
I'd like to know how can I get access to PX4 ulog for further debug.

Thank you.

Jaeyoung-Lim, 2019-08-13:
If you are running SITL, one approach(which I find quite convenient) you can try is to log from QGC and upload to flight review automatically.

mrivi, 2019-08-13:
The flight controller logs are in ` ~/.ros/log/ ` divided by date. So a log from today would be in `~/.ros/log/2019-08-13`

SwiftGust, 2019-08-13:
@Jaeyoung-Lim Thank you for sharing tip.But mavlink logging from QGC takes quite resources, causes lag on my desktop.

@mrivi I found the log in the exact path you mentioned, but it is little bit difficult to navigate, can we move SITL ulog to be stored under the directory somewhere like ~/.ros/px4_log. 
It is just a suggestion, thank you.