PX4/PX4-Avoidance #465: [Local Planner] Unable to takeoff after update

SwiftGust, 2019-08-13:
**Describe the bug**
With local planner, Misson mode immediately changes to hold mode with message
`WARN  [FlightTaskUtility] Obstacle Avoidance system failed, loitering`

However, offboard mode behavior is normal, can avoid obstacles with RViz goals
and also able to take-off with PX4 autopilot parameter COM_OBS_AVOID to 0

This started to happen with updating mavlink-gbp-release(release/kinetic/mavlink/2019.8.8-1) / mavros(v0.32.0 & master)  since PR #445 merged

I'm suspecting /mavros/companion_process/status heartbeat message does not taken by PX4 Firmware.
However, I have updated PX4 Firmware to the latest but also not working.
(https://github.com/PX4/Firmware/commit/106905871d374c125730fe675266d67c7f869469)

**To Reproduce**
Steps to reproduce the behavior:

Ubuntu 16.04 
ROS Kinetic
mavlink-gbp-release(release/kinetic/mavlink/2019.8.8-1) 
mavros(v0.32.0 & master)  
PX4 Firmware
(https://github.com/PX4/Firmware/commit/106905871d374c125730fe675266d67c7f869469)

Avoidance with PR #445 merged (https://github.com/PX4/avoidance/commit/d8499e1ee3cc89068830ee398d272104fd484dcd)

1. launch SITL with `roslaunch local_planner local_planner_depth-camera.launch`
2. Create & Upload any mission or execute take-off
3. Execute misson (if you did not execute take-off)

**Expected behavior**
A clear and concise description of what you expected to happen.
Flight mode immediately changes to hold while in ground.
with the message
`WARN  [FlightTaskUtility] Obstacle Avoidance system failed, loitering`

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=76a28146-6c82-4e05-b66b-b69c32894deb

Console output (from two separate launches) :
```
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /local_planner_node/goal_x_param: 17
 * /local_planner_node/goal_y_param: 15
 * /local_planner_node/goal_z_param: 3
 * /local_planner_node/pointcloud_topics: ['/camera/depth/p...
 * /local_planner_node/world_name: /home/uvify/backu...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
[... 394 lines cut ...]
[ WARN] [1565698357.508912998, 32.344000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698357.710241607, 32.544000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698357.713467406, 32.548000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698357.812905759, 32.644000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698357.817250016, 32.648000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698357.820487988, 32.652000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698358.011354975, 32.844000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698358.111677428, 32.944000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698358.114922858, 32.948000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698358.214135447, 33.044000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698358.218441645, 33.048000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698358.313671807, 33.144000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698358.317962287, 33.148000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698358.414403407, 33.244000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698358.616767342, 33.444000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698358.620021965, 33.448000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698358.815981342, 33.644000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698359.015682883, 33.844000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698359.019483387, 33.848000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698359.233566804, 34.044000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698359.433409473, 34.244000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698359.634810662, 34.444000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ WARN] [1565698359.734583128, 34.544000000]: TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived
[ INFO] [1565698359.737831446, 34.548000000]: FCU: DISARMED by Auto disarm on land
```

mrivi, 2019-08-13:
@SwiftGust You need to update to PX4 Firmware master such that you're using a more recent mavlink submodule. Something chnaged in the message we are using to send the setpoints. If you need to use 1.9 you need to use avoidance stable release 0.3.0

closed 2019-08-13 by mrivi

yxt916, 2020-09-11:
1.Check your environment varibale & path，put ALL models in the model path you set.
2.Check ros, is it installed fully？
3.Use a better computer.