PX4/PX4-Avoidance #471: TF Buffer error when launching local_planer

ngocdp, 2019-08-23:
Hi,

I follow the installation instruction in [PX4/avoidance](https://github.com/PX4/avoidance) and then I tried the local_planner by using:

`roslaunch local_planner local_planner_stereo.launch`

RVIZ opened up and I saw the tree, forest loaded up on RVIZ but not the drone.

When I look at the output of the Terminal i see a streamline of:
`
[ERROR] [1566528303.936276449, 3.004000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered`

The whole output are:
```
gijos@gijos-ThinkPad-T470:~$ roslaunch local_planner local_planner_stereo.launch 
... logging to /home/gijos/.ros/log/fd6bc512-c54f-11e9-8633-00216be8f31e/roslaunch-gijos-ThinkPad-T470-8948.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://gijos-ThinkPad-T470:35843/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /local_planner_node/goal_x_param: 15
 * /local_planner_node/goal_y_param: 15
 * /local_planner_node/goal_z_param: 4
 * /local_planner_node/pointcloud_topics: ['/stereo/points2']
 * /local_planner_node/world_name: /home/gijos/px4_a...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
[... 294 lines cut ...]
[ERROR] [1566528303.982168401, 3.040000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1566528303.985619541, 3.044000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1566528303.989883936, 3.048000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1566528303.994891954, 3.052000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1566528303.998127777, 3.056000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1566528304.001877437, 3.060000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1566528304.005026664, 3.064000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1566528304.008274562, 3.068000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1566528304.012488531, 3.072000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1566528304.016706536, 3.076000000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
```

My `ROS_PACKAGE_PATH`, `GAZEBO_MODEL_PATH`, `GAZEBO_PLUGIN_PATH` and `LD_LIBRARY_PATH`: 

```
ROS_PACKAGE_PATH /home/gijos/px4_avoidance/src/mavlink:/home/gijos/px4_avoidance/src/mavros/libmavconn:/home/gijos/px4_avoidance/src/mavros/mavros_msgs:/home/gijos/px4_avoidance/src/mavros/mavros:/home/gijos/px4_avoidance/src/mavros/mavros_extras:/home/gijos/px4_avoidance/src/avoidance/avoidance:/home/gijos/px4_avoidance/src/avoidance/global_planner:/home/gijos/px4_avoidance/src/avoidance/local_planner:/home/gijos/px4_avoidance/src/avoidance/safe_landing_planner:/home/gijos/px4_avoidance/src/mavros/test_mavros:/opt/ros/kinetic/share:/home/gijos/PX4/Firmware

GAZEBO_PLUGIN_PATH :/home/gijos/PX4/Firmware/build/px4_sitl_defaul/build_gazebo

GAZEBO_MODEL_PATH /home/gijos/.gazebo/models/:/home/gijos/PX4/Firmware/Tools/sitl_gazebo/models:/home/gijos/px4_avoidance/src/avoidance/avoidance/sim/models:/home/gijos/px4_avoidance/src/avoidance/avoidance/sim/worlds

LD_LIBRARY_PATH /home/gijos/px4_avoidance/devel/lib:/opt/ros/kinetic/lib:/opt/ros/kinetic/lib/x86_64-linux-gnu:/home/gijos/PX4/Firmware/build/px4_sitl_defaul/build_gazebo
```

For my first launch, it is just a warning, but from second launch, it turned to be an 'ERROR'

mrivi, 2019-08-23:
Hi @ngocdp, it seems that the communication between the Firmware and ROS is not up correctly. This is most likely due to a setup mistake. However the system variables you shared seem correct.
Are the system variables echoed from the same terminal where you tried to run the local planner? Could you please verify that the folder `/home/gijos/PX4/Firmware/build/px4_sitl_defaul/build_gazebo` actually exists? If you run `make px4_sitl gazebo` in the Firmware directory do you see the vehicle on the ground?

altshuler, 2019-08-25:
I have very similar issue.

I'm using :
Ubuntu 16.04
ROS Kinetic
mavros - compiled from master
mavlink - 2019.8.8-1
Firmware - v1.9.2

Here is the output:

```
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /local_planner_node/goal_x_param: 15
 * /local_planner_node/goal_y_param: 15
 * /local_planner_node/goal_z_param: 4
 * /local_planner_node/pointcloud_topics: ['/stereo/points2']
 * /local_planner_node/world_name: /home/evgeny/catk...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: local_origin
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: True
 * /mavros/local_position/tf/send_fcu: False
[... 399 lines cut ...]
WARN  [commander] Failsafe enabled: no RC and no offboard
[ERROR] [1566586059.952605468, 41.008000000]: FCU: Failsafe enabled: no RC and no offboard
[ INFO] [1566586060.064111874, 41.060000000]: FCU: Failsafe mode activated
[ INFO] [1566586062.163085563, 42.328000000]: FCU: Failsafe mode deactivated
WARN  [commander] Failsafe enabled: no RC and no offboard
[ERROR] [1566586063.630042888, 43.204000000]: FCU: Failsafe enabled: no RC and no offboard
[ INFO] [1566586063.726723353, 43.256000000]: FCU: Failsafe mode activated
[ INFO] [1566586065.455282379, 44.224000000]: FCU: Failsafe mode deactivated
WARN  [commander] Failsafe enabled: no RC and no offboard
[ERROR] [1566586066.999392709, 45.100000000]: FCU: Failsafe enabled: no RC and no offboard
[ INFO] [1566586067.119154599, 45.152000000]: FCU: Failsafe mode activated
[ INFO] [1566586068.908123461, 46.124000000]: FCU: Failsafe mode deactivated
WARN  [commander] Failsafe enabled: no RC and no offboard
[ERROR] [1566586070.339154001, 47.008000000]: FCU: Failsafe enabled: no RC and no offboard
[ INFO] [1566586070.418213978, 47.056000000]: FCU: Failsafe mode activated
[ INFO] [1566586072.047035233, 48.016000000]: FCU: Failsafe mode deactivated
WARN  [commander] Failsafe enabled: no RC and no offboard
[ERROR] [1566586073.513257001, 48.900000000]: FCU: Failsafe enabled: no RC and no offboard
[ INFO] [1566586073.593082549, 48.956000000]: FCU: Failsafe mode activated
[ INFO] [1566586075.402232564, 50.036000000]: FCU: Failsafe mode deactivated
WARN  [commander] Failsafe enabled: no RC and no offboard
[ERROR] [1566586076.884565970, 50.912000000]: FCU: Failsafe enabled: no RC and no offboard
[ INFO] [1566586076.991856182, 50.964000000]: FCU: Failsafe mode activated
[ INFO] [1566586079.046513695, 52.028000000]: FCU: Failsafe mode deactivated
```

ngocdp, 2019-08-26:
> Hi @ngocdp, it seems that the communication between the Firmware and ROS is not up correctly. This is most likely due to a setup mistake. However the system variables you shared seem correct.
> Are [...]

Hi @mrivi, thank you for your reply.

Are the system variables echoed from the same terminal where you tried to run the local planner?
- Yes, it is echoed in the same terminal that i used to launch the local_planner
Could you please verify that the folder `/home/gijos/PX4/Firmware/build/px4_sitl_defaul/build_gazebo` actually exists?
- Yes, it exists.
If you run `make px4_sitl gazebo` in the Firmware directory do you see the vehicle on the ground?
- No, I don't see the vehicle and I also got the following error:

```
make px4_sitl gazebo
[2/6] Performing configure step for 'sitl_gazebo'
FAILED: cd /home/gijos/PX4/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_VISION_ESTIMATION_DATA=ON -GNinja /home/gijos/PX4/Firmware/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/gijos/PX4/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
-- install-prefix: /usr/local
-- Using C++14 compiler
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   thread
--   filesystem
--   chrono
--   date_time
--   atomic
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   thread
--   system
--   filesystem
--   program_options
--   regex
--   iostreams
--   date_time
--   chrono
--   atomic
-- Boost version: 1.58.0
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Found Protobuf: /usr/local/lib/libprotobuf.a (Required is at least version "2.3.0") 
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.1.4
-- Checking for module 'uuid'
--   Found uuid, version 2.27.0
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 2.2.0
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- FreeImage.pc not found, we will search for FreeImage_INCLUDE_DIRS and FreeImage_LIBRARIES
-- Checking for module 'gts'
--   Found gts, version 0.7.6
-- Checking for module 'libswscale'
--   Found libswscale, version 3.1.101
-- Checking for module 'libavdevice >= 56.4.100'
--   Found libavdevice , version 56.4.100
-- Checking for module 'libavformat'
--   Found libavformat, version 56.40.101
-- Checking for module 'libavcodec'
--   Found libavcodec, version 56.60.100
-- Checking for module 'libavutil'
--   Found libavutil, version 54.31.100
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.7.2
-- Checking for module 'yaml-0.1'
--   Found yaml-0.1, version 0.1.6
-- Checking for module 'libzip'
--   Found libzip, version 1.0.1
-- Building klt_feature_tracker without catkin
-- Building OpticalFlow with OpenCV
-- catkin DISABLED
-- Found Protobuf: /usr/local/lib/libprotobuf.a  
Gazebo version: 9.10
CMake Error at CMakeLists.txt:415 (install):
  install DIRECTORY given non-directory
  "/home/gijos/PX4/Firmware/Tools/sitl_gazebo/models/README.md" to install.

-- Configuring incomplete, errors occurred!
See also "/home/gijos/PX4/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/gijos/PX4/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
ninja: build stopped: subcommand failed.
Makefile:200: recipe for target 'px4_sitl' failed
make: * [px4_sitl] Error 1
```
The two files CMakeError and CMakeOutput are attached below:
[CMakeError.log](https://github.com/PX4/avoidance/files/3539084/CMakeError.log)
[CMakeOutput.log](https://github.com/PX4/avoidance/files/3539085/CMakeOutput.log)

mrivi, 2019-08-26:
@ngocdp The firmware needs to be built for the avoidance to run correctly.
Are you running PX4 Firmware master or a release? Could you please make sure that your px4 toolchain is properly installed (you can find instructions here https://dev.px4.io/master/en/setup/dev_env_linux.html) ?

onionysy, 2019-09-17:
hi bro. i have similar question, too. But when i run make px4_sitl gazebo in the Firmware directory , i can see the vehicle on the ground.
`coamc@coamc-HP-Z840-Workstation:~/Firmware$ . ~/Firmware/Tools/setup_gazebo.bash ~/Firmware ~/Firmware/build/px4_sitl_default
GAZEBO_PLUGIN_PATH :/home/coamc/Firmware/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/coamc/catkin_ws/src/avoidance/avoidance/sim/models:/home/coamc/catkin_ws/src/avoidance/avoidance/sim/worlds:/home/coamc/catkin_ws/src/avoidance/avoidance/sim/models:/home/coamc/catkin_ws/src/avoidance/avoidance/sim/worlds:/home/coamc/Firmware/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /home/coamc/avoid_catkin_ws/devel/lib:/opt/ros/melodic/lib:/home/coamc/Firmware/build/px4_sitl_default/build_gazebo
coamc@coamc-HP-Z840-Workstation:~/Firmware$ export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:~/Firmware
coamc@coamc-HP-Z840-Workstation:~/Firmware$ echo export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:~/catkin_ws/src/avoidance/avoidance/sim/models:~/catkin_ws/src/avoidance/avoidance/sim/worlds >> ~/.bashrc
coamc@coamc-HP-Z840-Workstation:~/Firmware$ roslaunch local_planner local_planner_stereo.launch
... logging to /home/coamc/.ros/log/7e00678a-d8ee-11e9-8eef-44334c49fbe0/roslaunch-coamc-HP-Z840-Workstation-13248.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://coamc-HP-Z840-Workstation:43697/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /local_planner_node/goal_x_param: 15
 * /local_planner_node/goal_y_param: 15
 * /local_planner_node/goal_z_param: 4
 * /local_planner_node/pointcloud_topics: ['/stereo/points2']
 * /local_planner_node/world_name: /home/coamc/avoid...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: local_origin
[... 373 lines cut ...]
[ERROR] [1568685590.250653611]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1568685595.252252738]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1568685595.253554644]: PR: Unknown parameter to get: MPC_COL_PREV_D
[ERROR] [1568685595.254790184]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1568685595.256075273]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1568685600.257787692]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1568685600.259369207]: PR: Unknown parameter to get: MPC_COL_PREV_D
[ERROR] [1568685600.260613190]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1568685600.261845196]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1568685605.263504300]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1568685605.264800808]: PR: Unknown parameter to get: MPC_COL_PREV_D
[ERROR] [1568685605.266054103]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1568685605.267268010]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1568685610.268887227]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1568685610.270255759]: PR: Unknown parameter to get: MPC_COL_PREV_D
[ERROR] [1568685610.271465990]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1568685610.272592360]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1568685615.274354321]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1568685615.275883535]: PR: Unknown parameter to get: MPC_COL_PREV_D
[ERROR] [1568685615.277040675]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1568685615.278281762]: PR: Unknown parameter to get: NAV_ACC_RAD
`
it run the error, so i kill it.
i want to ask what's wrong. thank  @mrivi 
[image]

onionysy, 2019-09-17:
hi bro. i have similar question, too. But when i run make px4_sitl gazebo in the Firmware directory , i can see the vehicle on the ground.
`coamc@coamc-HP-Z840-Workstation:~/Firmware$ . ~/Firmware/Tools/setup_gazebo.bash ~/Firmware ~/Firmware/build/px4_sitl_default
GAZEBO_PLUGIN_PATH :/home/coamc/Firmware/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/coamc/catkin_ws/src/avoidance/avoidance/sim/models:/home/coamc/catkin_ws/src/avoidance/avoidance/sim/worlds:/home/coamc/catkin_ws/src/avoidance/avoidance/sim/models:/home/coamc/catkin_ws/src/avoidance/avoidance/sim/worlds:/home/coamc/Firmware/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /home/coamc/avoid_catkin_ws/devel/lib:/opt/ros/melodic/lib:/home/coamc/Firmware/build/px4_sitl_default/build_gazebo
coamc@coamc-HP-Z840-Workstation:~/Firmware$ export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:~/Firmware
coamc@coamc-HP-Z840-Workstation:~/Firmware$ echo export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:~/catkin_ws/src/avoidance/avoidance/sim/models:~/catkin_ws/src/avoidance/avoidance/sim/worlds >> ~/.bashrc
coamc@coamc-HP-Z840-Workstation:~/Firmware$ roslaunch local_planner local_planner_stereo.launch
... logging to /home/coamc/.ros/log/7e00678a-d8ee-11e9-8eef-44334c49fbe0/roslaunch-coamc-HP-Z840-Workstation-13248.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://coamc-HP-Z840-Workstation:43697/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /local_planner_node/goal_x_param: 15
 * /local_planner_node/goal_y_param: 15
 * /local_planner_node/goal_z_param: 4
 * /local_planner_node/pointcloud_topics: ['/stereo/points2']
 * /local_planner_node/world_name: /home/coamc/avoid...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: local_origin
[... 375 lines cut ...]
[ERROR] [1568685595.253554644]: PR: Unknown parameter to get: MPC_COL_PREV_D
[ERROR] [1568685595.254790184]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1568685595.256075273]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1568685600.257787692]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1568685600.259369207]: PR: Unknown parameter to get: MPC_COL_PREV_D
[ERROR] [1568685600.260613190]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1568685600.261845196]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1568685605.263504300]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1568685605.264800808]: PR: Unknown parameter to get: MPC_COL_PREV_D
[ERROR] [1568685605.266054103]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1568685605.267268010]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1568685610.268887227]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1568685610.270255759]: PR: Unknown parameter to get: MPC_COL_PREV_D
[ERROR] [1568685610.271465990]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1568685610.272592360]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1568685615.274354321]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1568685615.275883535]: PR: Unknown parameter to get: MPC_COL_PREV_D
[ERROR] [1568685615.277040675]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1568685615.278281762]: PR: Unknown parameter to get: NAV_ACC_RAD
`
it run the error, so i kill it.
i want to ask what's wrong. thank you!
[image]

make px4_sitl_default gazebo

onionysy, 2019-09-17:
@mrivi

onionysy, 2019-09-17:
i try again and it change to this
`coamc@coamc-HP-Z840-Workstation:~/Firmware$ roslaunch local_planner local_planner_stereo.launch
... logging to /home/coamc/.ros/log/61dd7502-d8f3-11e9-8eef-44334c49fbe0/roslaunch-coamc-HP-Z840-Workstation-15412.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://coamc-HP-Z840-Workstation:39483/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /local_planner_node/goal_x_param: 15
 * /local_planner_node/goal_y_param: 15
 * /local_planner_node/goal_z_param: 4
 * /local_planner_node/pointcloud_topics: ['/stereo/points2']
 * /local_planner_node/world_name: /home/coamc/avoid...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: local_origin
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: True
 * /mavros/local_position/tf/send_fcu: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: False
[... 188 lines cut ...]
[ WARN] [1568687546.762324140]: TM : RTT too high for timesync: 18446743543173.55 ms.
[ WARN] [1568687547.364918485]: TM : RTT too high for timesync: 18446743542573.55 ms.
[ WARN] [1568687547.966004495]: TM : RTT too high for timesync: 18446743541973.55 ms.
[ WARN] [1568687548.566392070]: TM : RTT too high for timesync: 18446743541373.55 ms.
[ WARN] [1568687549.166364958]: TM : RTT too high for timesync: 18446743540773.55 ms.
[ WARN] [1568687549.766566951]: TM : RTT too high for timesync: 18446743540173.55 ms.
[ WARN] [1568687550.366403711]: TM : RTT too high for timesync: 18446743539573.55 ms.
[ERROR] [1568687550.392129298]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1568687550.393856488]: PR: Unknown parameter to get: MPC_COL_PREV_D
[ERROR] [1568687550.395527875]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1568687550.397087052]: PR: Unknown parameter to get: NAV_ACC_RAD
[ WARN] [1568687550.966317149]: TM : RTT too high for timesync: 18446743538973.55 ms.
[ WARN] [1568687551.569440052]: TM : RTT too high for timesync: 18446743538373.55 ms.
[ WARN] [1568687552.170070225]: TM : RTT too high for timesync: 18446743537773.55 ms.
[ WARN] [1568687552.770034684]: TM : RTT too high for timesync: 18446743537173.55 ms.
[ WARN] [1568687553.370531860]: TM : RTT too high for timesync: 18446743536573.55 ms.
[ WARN] [1568687553.970412392]: TM : RTT too high for timesync: 18446743535973.55 ms.
[ WARN] [1568687554.570737686]: TM : RTT too high for timesync: 18446743535373.55 ms.
[ WARN] [1568687555.170559939]: TM : RTT too high for timesync: 18446743534773.55 ms.
[ERROR] [1568687555.399407232]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1568687555.401084783]: PR: Unknown parameter to get: MPC_COL_PREV_D
[ERROR] [1568687555.402661459]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1568687555.404010770]: PR: Unknown parameter to get: NAV_ACC_RAD
`
i don't know why! how can i fix it

mrivi, 2019-10-21:
fixed

closed 2019-10-21 by mrivi

Cristian-wp, 2020-05-26:
Excuse me, how was fixed?

yxt916, 2020-09-06:
1000000 TF Buffer errors, help!

yxt916, 2020-09-11:
1.Check your environment varibale & path，put ALL models in the model path you set.
2.Check ros, is it installed fully？
3.Use a better computer.
99% is model path wrong or model misssing or computer too old.

tiger-k, 2022-05-07:
I use a new px4_config.yaml just copy from avoidancv/avoidance/resourec/px4_congih.yaml.
and then this error disappeared.