PX4/PX4-Avoidance #473: TF Buffer: tf has already been dropped from buffer

tomasgareau, 2019-08-27:
Running the local planner on hardware with:
- Intel Aero RTF Drone
- Ubuntu **16.04.6 LTS**
- Mavros **v0.32.1-1xenial-20190813-133606**
- PX4 Firmware **v1.9.2** (downloaded from release in PX4/Firmware)
- Realsense R200 Camera (FW version **1.0.71.06**)
- Local Planner Node **v0.3.0** (built from 0.3.0 tag in this repo)

produces a constant stream of

`TF Buffer: could not retrieve requested transform from buffer, tf has not yet arrived`

messages when trying to get the transform from /camera_depth_optical_frame to /local_origin. 

When testing collision prevention in Position Mode, the drone does not react to obstacles within the MPC_COL_PREV_D of 1.5 metres, even at slow speeds.

**Additional Debug Info**
_local pointcloud_
/local_pointcloud is being published at 4-6 Hz but the installation instructions mention that 10-20 Hz is expected... is this indicative of some issue in my setup?

_tf monitor_
Running tf_monitor seems to suggest that the camera_depth_optical_frame -> local_origin transform is being published regularly:

```
RESULTS: for camera_depth_optical_frame to local_origin
Chain is: camera_depth_optical_frame -> camera_depth_frame -> camera_link -> fcu -> local_origin
Net delay      avg = 0.026561: max = 0.0535834

Frames:
Frame: camera_depth_frame published by unknown_publisher(static) Average Delay: 0 Max Delay: 0
Frame: camera_depth_optical_frame published by unknown_publisher(static) Average Delay: 0 Max Delay: 0
Frame: camera_link published by unknown_publisher Average Delay: -0.00935239 Max Delay: 0
Frame: fcu published by unknown_publisher Average Delay: 0.013005 Max Delay: 0.0219217

All Broadcasters:
Node: unknown_publisher 158.796 Hz, Average Delay: -0.00102539 Max Delay: 0.0219217
Node: unknown_publisher(static) 41943 Hz, Average Delay: 0 Max Delay: 0
```

_rosout log_
[tf_buffer_issue_rosout.log](https://github.com/PX4/avoidance/files/3546818/tf_buffer_issue_rosout.log)

mrivi, 2019-08-29:
Could you check the rate at which you're getting /camera/depth/points and /mavros/local_position/pose?

tomasgareau, 2019-08-29:
/mavros/local_position/pose seems to get a steady ~30 Hz update:

```
rostopic hz /mavros/local_position/pose
subscribed to [/mavros/local_position/pose]
average rate: 29.839
              min: 0.023s max: 0.042s std dev: 0.00410s window: 29
average rate: 29.536
              min: 0.023s max: 0.064s std dev: 0.00550s window: 58
```

and /camera/depth/points typically between ~24-27 Hz:

```
rostopic hz /camera/depth/points
subscribed to [/camera/depth/points]
average rate: 27.807
        min: 0.030s max: 0.069s std dev: 0.00921s window: 26
average rate: 27.365
        min: 0.030s max: 0.069s std dev: 0.00980s window: 53
average rate: 26.251
        min: 0.030s max: 0.069s std dev: 0.01138s window: 78
average rate: 25.876
        min: 0.030s max: 0.069s std dev: 0.01203s window: 102
average rate: 25.763
        min: 0.030s max: 0.069s std dev: 0.01228s window: 128
```

mrivi, 2019-08-29:
@tomasgareau  we're seeing those warnings as well and we're investigating why. However the planner is running fast enough to do collision prevention. Have you tried increasing `MPC_COL_PREV_D` to 5/6 meters?

tomasgareau, 2019-08-30:
Increasing `MPC_COL_PREV_D` to 5 metres did the trick to get collision prevention working in Position Mode, thanks!

I can close this issue now unless you'd like to keep it up to track the warning messages until those are resolved.

mrivi, 2019-10-21:
fixed

closed 2019-10-21 by mrivi