PX4/PX4-Avoidance #480: Stuck in ALTITUDE_CHANGE state after takeoff

stevegray954, 2019-09-09:
SITL quadrotor stuck in ALTITUDE CHANGE state after going into offboard mode, arming, and taking off, as of commit 2b1d600.

Issue seems to be caused by this snippet in WaypointGenerator::isAltitudeChange() in [waypoint_generator.cpp](https://github.com/PX4/avoidance/blob/master/local_planner/src/nodes/waypoint_generator.cpp) -- the `return true;` should really be below the `ROS_INFO` line.  Otherwise, the code is unable to reach the only location that could set `reach_altitude_offboard` to true.

```
  if (need_to_change_altitude) {
    return true;

    if (nav_state_ == NavigationState::offboard) {
      if (position_.z() > goal_.z()) {
        reach_altitude_offboard_ = true;
        return false;
      }
    }

    ROS_INFO("\033[1;35m[OA] Reach height first \033[0m");
  }
```

mrivi, 2019-09-12:
I am going to look into this. Thanks for reporting

mrivi, 2019-09-12:
See PR #481

closed 2019-09-19 by mrivi