PX4/PX4-Avoidance #485: issue to start local_planner

onionysy, 2019-09-17:
We use GitHub issues only to discuss PX4 Avoidance bugs and new features. For
questions about using PX4 or related components, please use [PX4 Discuss](http://discuss.px4.io/).
 
i try to roslaunch local_planner local_planner_stereo.launch again and it change to this
`coamc@coamc-HP-Z840-Workstation:~/Firmware$ roslaunch local_planner local_planner_stereo.launch
... logging to /home/coamc/.ros/log/61dd7502-d8f3-11e9-8eef-44334c49fbe0/roslaunch-coamc-HP-Z840-Workstation-15412.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://coamc-HP-Z840-Workstation:39483/
SUMMARY

CLEAR PARAMETERS

    /mavros/

PARAMETERS

    /gazebo/enable_ros_network: True
    /local_planner_node/goal_x_param: 15
    /local_planner_node/goal_y_param: 15
    /local_planner_node/goal_z_param: 4
    /local_planner_node/pointcloud_topics: ['/stereo/points2']
    /local_planner_node/world_name: /home/coamc/avoid...
    /mavros/cmd/use_comp_id_system_control: False
    /mavros/conn/heartbeat_rate: 1.0
    /mavros/conn/system_time_rate: 1.0
    /mavros/conn/timeout: 10.0
    /mavros/conn/timesync_rate: 10.0
    /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
    /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
    /mavros/distance_sensor/hrlv_ez4_pub/id: 0
    /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
    /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
    /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
    /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
    /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
    /mavros/distance_sensor/laser_1_sub/id: 3
    /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
    /mavros/distance_sensor/laser_1_sub/subscriber: True
    /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
    /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
    /mavros/distance_sensor/lidarlite_pub/id: 1
    /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
    /mavros/distance_sensor/lidarlite_pub/send_tf: True
    /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
    /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
    /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
    /mavros/distance_sensor/sonar_1_sub/id: 2
    /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
    /mavros/distance_sensor/sonar_1_sub/subscriber: True
    /mavros/fcu_protocol: v2.0
    /mavros/fcu_url: udp://:14540@loca...
    /mavros/gcs_url:
    /mavros/global_position/frame_id: fcu
    /mavros/global_position/rot_covariance: 99999.0
    /mavros/global_position/tf/child_frame_id: fcu_utm
    /mavros/global_position/tf/frame_id: local_origin
    /mavros/global_position/tf/send: False
    /mavros/image/frame_id: px4flow
    /mavros/imu/angular_velocity_stdev: 0.000349065850399
    /mavros/imu/frame_id: fcu
    /mavros/imu/linear_acceleration_stdev: 0.0003
    /mavros/imu/magnetic_stdev: 0.0
    /mavros/imu/orientation_stdev: 1.0
    /mavros/local_position/frame_id: local_origin
    /mavros/local_position/tf/child_frame_id: fcu
    /mavros/local_position/tf/frame_id: local_origin
    /mavros/local_position/tf/send: True
[... 195 lines cut ...]
[ WARN] [1568687549.166364958]: TM : RTT too high for timesync: 18446743540773.55 ms.
[ WARN] [1568687549.766566951]: TM : RTT too high for timesync: 18446743540173.55 ms.
[ WARN] [1568687550.366403711]: TM : RTT too high for timesync: 18446743539573.55 ms.
[ERROR] [1568687550.392129298]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1568687550.393856488]: PR: Unknown parameter to get: MPC_COL_PREV_D
[ERROR] [1568687550.395527875]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1568687550.397087052]: PR: Unknown parameter to get: NAV_ACC_RAD
[ WARN] [1568687550.966317149]: TM : RTT too high for timesync: 18446743538973.55 ms.
[ WARN] [1568687551.569440052]: TM : RTT too high for timesync: 18446743538373.55 ms.
[ WARN] [1568687552.170070225]: TM : RTT too high for timesync: 18446743537773.55 ms.
[ WARN] [1568687552.770034684]: TM : RTT too high for timesync: 18446743537173.55 ms.
[ WARN] [1568687553.370531860]: TM : RTT too high for timesync: 18446743536573.55 ms.
[ WARN] [1568687553.970412392]: TM : RTT too high for timesync: 18446743535973.55 ms.
[ WARN] [1568687554.570737686]: TM : RTT too high for timesync: 18446743535373.55 ms.
[ WARN] [1568687555.170559939]: TM : RTT too high for timesync: 18446743534773.55 ms.
[ERROR] [1568687555.399407232]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1568687555.401084783]: PR: Unknown parameter to get: MPC_COL_PREV_D
[ERROR] [1568687555.402661459]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1568687555.404010770]: PR: Unknown parameter to get: NAV_ACC_RAD
`
i don't know why! how can i fix it @mrivi
can you give me some advices.
thank you

Thanks!

mrivi, 2019-09-17:
```bash
[gazebo-6] process has died [pid 15452, exit code 255, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -e ode /home/coamc/avoid_catkin_ws/src/avoidance/avoidance/sim/worlds/simple_obstacle.world __name:=gazebo __log:=/home/coamc/.ros/log/61dd7502-d8f3-11e9-8eef-44334c49fbe0/gazebo-6.log].
log file: /home/coamc/.ros/log/61dd7502-d8f3-11e9-8eef-44334c49fbe0/gazebo-6*.log

Gazebo died. Does that always happen? It can happen sometimes that Gazebo doesn't start up properly and you need to try a couple of times.

onionysy, 2019-09-17:
> ```shell
> [gazebo-6] process has died [pid 15452, exit code 255, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -e ode /home/coamc/avoid_catkin_ws/src/avoidance/avoidance/sim/worlds/simple_obstacle.world __name:=gazebo __log:=/home/coamc/.ros/log/61dd7502-d8f3-11e9-8eef-44334c49fbe0/gazebo-6.log].
> log file: /home/coamc/.ros/log/61dd7502-d8f3-11e9-8eef-44334c49fbe0/gazebo-6*.log
> ```
> 
> Gazebo died. Does that always happen? It can happen sometimes that Gazebo doesn't start up properly and you need to try a couple of times.

i wonder  after i    make px4_sitl_default gazebo   in   build and run the simulation    step
should i kill the gazebo simulation in terminal?

mrivi, 2019-09-17:
yes, you need to run `make px4_sitl gazebo` then kill the simulation. Afterwards you can run `roslaunch local_planner local_planner_depth-camera.launch`

onionysy, 2019-09-17:
> yes, you need to run `make px4_sitl gazebo` then kill the simulation. Afterwards you can run `roslaunch local_planner local_planner_depth-camera.launch`

i try it again and it shows this
`coamc@coamc-HP-Z840-Workstation:~/Firmware$ roslaunch local_planner local_planner_stereo.launch
... logging to /home/coamc/.ros/log/f306d5c8-d90c-11e9-8eef-44334c49fbe0/roslaunch-coamc-HP-Z840-Workstation-7072.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://coamc-HP-Z840-Workstation:35861/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /local_planner_node/goal_x_param: 15
 * /local_planner_node/goal_y_param: 15
 * /local_planner_node/goal_z_param: 4
 * /local_planner_node/pointcloud_topics: ['/stereo/points2']
 * /local_planner_node/world_name: /home/coamc/avoid...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: local_origin
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: True
 * /mavros/local_position/tf/send_fcu: False
[... 234 lines cut ...]
INFO  [px4_work_queue] creating: wq:hp_default, priority: 87, stack: 16384 bytes
INFO  [px4_work_queue] creating: wq:att_pos_ctrl, priority: 88, stack: 16384 bytes
INFO  [px4_work_queue] creating: wq:rate_ctrl, priority: 99, stack: 16384 bytes
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14558 remote port 14541
INFO  [logger] logger started (mode=all)
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [logger] Start file log (type: full)
[ INFO] [1568700987.455891109]: udp0: Remote address: 127.0.0.1:14580
INFO  [logger] Opened full log file: ./log/2019-09-17/06_16_27.ulg
[ INFO] [1568700987.456421493]: CON: Got HEARTBEAT, connected. FCU: PX4 Autopilot
INFO  [px4] Startup script returned successfully
[ INFO] [1568700987.458185285]: FCU: [logger] file: ./log/2019-09-17/06_16_27.ulg
INFO  [mavlink] partner IP: 127.0.0.1
[ERROR] [1568700991.287621926]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1568700991.288987250]: PR: Unknown parameter to get: MPC_COL_PREV_D
[ERROR] [1568700991.290099845]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1568700991.290996042]: PR: Unknown parameter to get: NAV_ACC_RAD
[ERROR] [1568700996.292613211]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1568700996.293569398]: PR: Unknown parameter to get: MPC_COL_PREV_D
[ERROR] [1568700996.294475101]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1568700996.295387345]: PR: Unknown parameter to get: NAV_ACC_RAD
`

mrivi, 2019-09-17:
The first time you load the world it takes quite a bit of time, that's why you see those unknown parameters.
You can change [this parameter](https://github.com/PX4/avoidance/blob/master/avoidance/launch/avoidance_sitl_mavros.launch#L3) to true such that you see when the Gazebo world has loaded.

onionysy, 2019-09-17:
thanks for your guide. i succed in lauch the gazebo and get map in the rviz.
However, i want the uav to the target point but the UAV could only change its altitude but not change its x,y position
i runin offboard mode
can you give me some advices. thanks a lot
`coamc@coamc-HP-Z840-Workstation:~/Firmware$ roslaunch local_planner local_planner_stereo.launch
... logging to /home/coamc/.ros/log/f306d5c8-d90c-11e9-8eef-44334c49fbe0/roslaunch-coamc-HP-Z840-Workstation-10952.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://coamc-HP-Z840-Workstation:45733/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /local_planner_node/goal_x_param: 15
 * /local_planner_node/goal_y_param: 15
 * /local_planner_node/goal_z_param: 4
 * /local_planner_node/pointcloud_topics: ['/stereo/points2']
 * /local_planner_node/world_name: /home/coamc/avoid...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: local_origin
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: True
[... 275 lines cut ...]
[ERROR] [1568704959.435037891, 1.260000000]: Ignoring transform for child_frame_id "fcu" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 0.000000 0.000000 0.000000)
Error:   TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "fcu" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 0.000000 0.000000 0.000000)
         at line 257 in /tmp/binarydeb/ros-melodic-tf2-0.6.5/src/buffer_core.cpp
[ERROR] [1568704959.435167602, 1.260000000]: Ignoring transform for child_frame_id "fcu" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 0.000000 0.000000 0.000000)
[ INFO] [1568704960.205307526, 2.028000000]: Successfully loaded rviz world
[ERROR] [1568704960.415506240, 2.236000000]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1568704960.416781108, 2.236000000]: PR: Unknown parameter to get: MPC_COL_PREV_D
[ERROR] [1568704960.418256945, 2.240000000]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1568704960.419422434, 2.240000000]: PR: Unknown parameter to get: NAV_ACC_RAD
INFO  [ecl/EKF] EKF commencing GPS fusion
[ERROR] [1568704965.421052651, 7.236000000]: PR: Unknown parameter to get: MPC_XY_CRUISE
[ERROR] [1568704965.422371469, 7.240000000]: PR: Unknown parameter to get: MPC_COL_PREV_D
[ERROR] [1568704965.423657021, 7.240000000]: PR: Unknown parameter to get: MPC_LAND_SPEED
[ERROR] [1568704965.424906018, 7.240000000]: PR: Unknown parameter to get: NAV_ACC_RAD
[ INFO] [1568704973.190613025, 15.000000000]: WP: mission received
[ INFO] [1568704984.340393510, 26.140000000]: Setting goal: Frame:local_origin, Position(35.594, -1.335, 0.000), Orientation(0.000, 0.000, -0.013, 1.000) = Angle: -0.027

[ INFO] [1568704992.508392452, 34.296000000]: Setting goal: Frame:local_origin, Position(55.649, 0.967, 0.000), Orientation(0.000, 0.000, 1.000, 0.014) = Angle: 3.113

[ WARN] [1568705003.720541922, 45.500000000]: CMD: Unexpected command 176, result 0
[ INFO] [1568705008.239756335, 50.016000000]: FCU: ARMED by Arm/Disarm component command
INFO  [commander] Takeoff detected
[ INFO] [1568705009.450282063, 51.224000000]: FCU: Takeoff detected

`

onionysy, 2019-09-17:
[image]

mrivi, 2019-09-17:
yes, there is known bug in offboard. You can either use [this branch](https://github.com/PX4/avoidance/pull/481) for offboard control or you can use the master branch in Mission Mode.

onionysy, 2019-09-18:
thanks for your guide!i have solve my problem.

closed 2019-09-18 by mrivi