PX4/PX4-Avoidance #491: [Local Planner] Notable off-course flight during survey mission

SwiftGust, 2019-09-30:
I'm trying survey mission with local planner in Gazebo SITL and found local planner causes severe off-course during survey mission without any obstacle points are near, which is highly notable during new waypoint position is at 90 degree off from current heading like picture below
(notable in WP 9 - 10, 15 - 16, 21 - 22  ...)

[image]

disable smoothing_speed_xy_ to zero makes less notable this issue but still causing off-course flight

Gazebo world - simple_obstacle.world (Default, VRC terrain)

PX4 Firmware version : https://github.com/PX4/Firmware/commit/9bc19fd7328ad1a44d1a87c5e73cdb6392bbbfe4

avoidance version : https://github.com/PX4/avoidance/commit/79d2c8b797e376bbc0d2eb517dfb36d703af9197

Mission file : 
[Survey-SITL.zip](https://github.com/PX4/avoidance/files/3669096/Survey-SITL.zip)

mrivi, 2019-09-30:
Flew your plan and I am not seeing this crazy overshoot. Do you have the flight controller log? Which MPC_AUTO_MODE are you using? Cruise speed?
[image]

SwiftGust, 2019-09-30:
I missed point, I made cruise speed set to 5m/s but it seems like did not applied in exported mission.
In cruise speed of 2m/s it was fine to me too. 

MPC_AUTO_MODE = 1 (Jerk-limited trajectory)
And PX4 side is well-tuned, It follows almost same as planned course if COM_OBS_AVOID = 0.

mrivi, 2019-09-30:
@SwiftGust - please set MPC_AUTO_MODE to 0. There is an interaction between the control loop in the planner and in the Firmware with Jerk Limited Trajectory. 
The local planner is currently being tested and developed with  MPC_AUTO_MODE=0. You will see that even at 5m/s the planner is able to follow the survey trajectories nicely.

SwiftGust, 2019-09-30:
@mrivi setting MPC_AUTO_MODE to 0, Now the vehicle follows the course very nice & closely.

Thank you :+1:  

[image]

closed 2019-09-30 by SwiftGust