PX4/PX4-Avoidance #492: Invalid Quaternion in transform at startup

mrivi, 2019-09-30:
We get a bunch of this errors at startup on all 3 planners. We need to figure out where are they coming from. 
``` [ERROR] [1569826119.454300726, 0.856000000]: Ignoring transform for child_frame_id "fcu" from authority "unknown_publisher" because of an invalid quaternion in the transform (0.000000 0.000000 0.000000 0.000000)
```

mrivi, 2019-10-01:
I tried a few things:
- happens also with older versions of mavros e.g. 0.30.0
- doesn't happen on hardware

hassaanhashmi, 2019-10-14:
Hi @mrivi , have you figured out why it happens in software?

mrivi, 2019-10-14:
No, I didn't investigate further since they aren't creating any issues. We look further in the future.

mrivi, 2019-10-14:
Figured out where it comes ffrom https://github.com/PX4/Firmware/issues/13182

mrivi, 2019-10-15:
Fixed https://github.com/PX4/Firmware/pull/13184

closed 2019-10-15 by mrivi