PX4/PX4-Avoidance #493: global planner node died

JoeyLeeNPU, 2019-10-01:
In recent days, I update mavros and px4 to the lastest version, but when I launch the global planner node on my TX2, the ternimal will show that the global planner node is died. The error is about **std::bad_alloc**, at the same time, my TX2's memory is increasing so quickly.
I use the same launch file and parameters to run this node on my laptop is normal without any problem, and memory is stable.
In addition, I have run this node on my TX2 two months ago, it works very well. 

Any advice? Thanks for help!
Joey

Jaeyoung-Lim, 2019-10-01:
@JoeyLeeNPU Which version of avoidance are you using?

JoeyLeeNPU, 2019-10-01:
version is **Master**, the problem appears in this version. I remember the version that the mutex added for solving another global planner node died can work. @Jaeyoung-Lim

Jaeyoung-Lim, 2019-10-01:
@JoeyLeeNPU Just to be clear, do you mean master branch of today? It might be possible that you have different versions on your laptop and TX2

JoeyLeeNPU, 2019-10-01:
I confirm that both of my laptop and TX2 is version master. I also tried that copy the code from my laptop to tx2, but it have the same problem, I git the code to TX2, problem is also occured. @Jaeyoung-Lim

Jaeyoung-Lim, 2019-10-01:
@JoeyLeeNPU master is the name of the branch and not the version. Could you check the hash of the avoidance on TX2 and your laptop?

JoeyLeeNPU, 2019-10-02:
Could U tell me how to check the hash of the avoidance, thanks~ @Jaeyoung-Lim

JoeyLeeNPU, 2019-10-02:
The hash of the avoidance on both laptop and TX2 is **c38bc8f200c209ab36785eae8283e616680aa98a**.
Thanks!

Joey

Jaeyoung-Lim, 2019-10-02:
@JoeyLeeNPU I just noticed that you said you are running the same launch file as on your laptop. Are you trying to run SITL on the TX2? 

If you are trying to fly a vehicle, you need to have a proper launchfile for your vehicle

JoeyLeeNPU, 2019-10-02:
I will use another TX2 to try avoidance, and update the progress in time.

Jaeyoung-Lim, 2019-10-02:
@JoeyLeeNPU What do you mean using another TX2? You are simply using the wrong launchfile on the TX2

JoeyLeeNPU, 2019-10-02:
I write a new launchfile for my drone, and it disables SITL. And the same launchfile can work on my TX2 two months ago. I think my problem is not occured because of the wrong launchfile.

JoeyLeeNPU, 2019-10-04:
@Jaeyoung-Lim I tried the same code on my another TX2, and we met the same problem as following,
[image]
but I use another x86 processor to run avoidacne, it has no problem.
Hence, could U try the avoidance on the arm-structure procossor such as tx2?

Thanks! 😄
Joey

Jaeyoung-Lim, 2019-10-04:
I think @SwiftGust might be able to help. I believe he has some experience with running the global planner on the TX2

JoeyLeeNPU, 2019-10-04:
@SwiftGust Hi~ I want to use avoidance on my tx2 in order to navigate my drone, and I use two TF, one is to get pose from /map to /baselink (because i got my pose from SLAM), another is to get /local-ned to baselink. I use the second TF for avoidance. 
Now i met a problem when i run global planner such as the last comment shows. could u help me?

Joey

SwiftGust, 2019-10-04:
@JoeyLeeNPU Hi, I never met this problem before, and I'm using older version of the global planner.

Maybe you can try out some things, `std::bad_alloc` usually coming from using dynamic allocation with `new` keyword 

1. `Successfully loaded rviz world` 
this message is coming from Rviz loader which is merely for SITL, can you build and try again with `-DDISABLE_SIMUALTION=ON`

2. You may need to debug your global planner inside Jetson TX2 using GDB or try inserting some additional code for debug to find out what is really causing the issue or try adding exception handler for more informative message

For attaching debugger, refer - http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB

JoeyLeeNPU, 2019-10-04:
@SwiftGust , thanks for your response.
I remember global plnanner fix the problem by adding a **mutex** . Your version is before this or later?
If can, could u send me your open-source code to me, I want to refer your code to check my code.

Joey

SwiftGust, 2019-10-04:
@JoeyLeeNPU Sorry but I cannot provide code, version is way before mutex included.

JoeyLeeNPU, 2019-10-04:
@SwiftGust I met this problem when i update the version fixed mutex.  Thanks for your help!
@Jaeyoung-Lim I will use the avoidance on my x86-structure procoessor first.

Joey

JoeyLeeNPU, 2019-10-05:
@Jaeyoung-Lim I found the point cloud resolution is too high for TX2 to calculate octomap, I set VGA on zed, and then global planner works well.

closed 2019-10-05 by JoeyLeeNPU