PX4/PX4-Avoidance #499: Parameters setting for indoor navigation tasks

JoeyLeeNPU, 2019-10-07:
I am preparing a indoor uav navigation task, in order to achieve collision-free flight i use avoidance. When I set the start position is (0,0,1.5), then drone can take off and hover at 1.5m and wait for next destination, next I choose a point free on octomap whose altitude is 1.5m, it always shows that the goal is occupied. My parameter is set as cell_scale=0.5, octomap resolution=0.5, goal_altitude=1.5, min_altitude=0, max_altitude=3, ground_obstacles_filter=0.

In addition, for indoor navigation, our drone is not very large, so above parameters may be rough. Could I set the cell_scale less than 0.5. and decrease the octomap's resolution, or some other suggestions for narrow obstacles avoidance?

Thanks,
Joey

Jaeyoung-Lim, 2019-10-07:
@JoeyLeeNPU The current `global_planner` is not usable for indoor navigation, as it uses priors of the occupancy probability of altitude (it is less probable to be occupied as the altitude increases). Therefore, it is likely that you will have a collision with the ceiling. 

A more robust planner is being developed on #453

JoeyLeeNPU, 2019-10-08:
hah! I see #453 and it may be a better solution for indoor navigation, but I also want to have a try on traditional global planner in avoidance.
In a room, I try to set **cell_scale** in source file as 0.5 and set the octomap resolution as 0.25 in debug window, changing parameter **cell_scale**  may not work, so I met that no matter how I change it, I can't get the desired value. Finally, I change it in the source code file.
For altitude priors, I set the altitude below 3m equal, and increase the cost of rising.
If OMPL added into avoidance, i will try it as soon as possible.

Thanks for help,
Joey

Jaeyoung-Lim, 2019-10-08:
@JoeyLeeNPU Awesome. I have seen some implementations working indoors with the changed priors, and seems to work with some modifications.

It would be cool if you can contribute the changes so that everyone else can also use it indoors

JoeyLeeNPU, 2019-10-08:
@Jaeyoung-Lim Okay, I will make some changes to meet both indoor and outdoor obstacle avoidance flights. If a lot of tests are successful, I will share my experience.

Cheers,
Joey

closed 2019-10-09 by JoeyLeeNPU