PX4/PX4-Avoidance #514: Maybe two bugs in global planner and the questions about planner

JoeyLeeNPU, 2019-10-22:
@Jaeyoung-Lim I am back again hah~
The bugs reported：
1. In cell.cpp line 70, the fucntion **ceil** may should use "floor", because when the radius is 3, the z-pos is wrong due to root a negative number.
2. The radius should be equal "**robot_radius/cell_scale**", rather than "**robot_radius/resolution**".

The question about global planner:
1. In avoidance, when I set a target point in rviz, it will judge if there is free. If there is free, the drone will fly to it, otherwise, the drone will hover. But I think in the actual flight, the octomap may not be always accurate for judging a location whether is free or not. 
When the target position is far away from the drone, it should fly closed to the target position, if there is not free, it should hover at the closest position to the target position.
2. Global planner will stop working when it plans some times without paths results, that is, the goal is free, but the drone is hover. Is there a problem with the parameter settings?

Thanks for help!
Joey

Jaeyoung-Lim, 2019-11-14:
@JoeyLeeNPU Sorry for the late reply. 

1. I don't quite get what you mean. Could you clarify on what you mean?
2. I believe the cell scale is always 1 and is not really used.

Global Planner
1. You are correct. And it will be only makred as occupied if it has "seen" it occupied. So if the target is far away it won't try to predict but rather always mark it as free as it has never been observed. It is a optimistic planner in that sense
2. I agree that the logic is not great. It will simply hover if it couldn't find a path. There is no path exploring logic embedded into global planner and that is why it will simply stay there if it couldn't find a path. I am planning to implement this kind of exploratory behavior, maybe something similar to the NBVP planner but maybe on the OMPL implementation

JoeyLeeNPU, 2019-12-02:
Sorry for too late response, my first question is that in  /global planner/cell.cpp line 70, the function "ceil" may should use "floor", if use "ceil" the z-pos will be calculated as a wrong number because of a square root of a negative number.

And now I use the local planner, I met a crash that when I publish a goal quickly in two times, the drone will be fly straightly in a high speed. I found that in **/local planner/src/nodes/local_planner.cpp** **line 124**, when the previous_goal is equal to goal, the normalization of them may be a NAN.

baumanta, 2019-12-10:
@JoeyLeeNPU I just found that bug as well in a different setting. If you do RTL twice in a row you also end up with prev_goal = goal. 
I fixed it here: https://github.com/PX4/avoidance/pull/541

JoeyLeeNPU, 2019-12-10:
@baumanta thanks for fixing this bug!

closed 2019-12-10 by JoeyLeeNPU