PX4/PX4-Avoidance #518: Local planner kills rviz when landing

SwiftGust, 2019-10-24:
**Describe the bug**
local planner kills rviz when starts to landing
probably due to something wrong in the message which does not properly handled in RViz

**To Reproduce**
Steps to reproduce the behavior:
1. Launch local planner SITL
2. Take-off the vehice
3. Initiate land mode or RTL (allow it to land with non-negative value in RTL_LAND_DELAY)
4. See rviz crashes

**Expected behavior**
local_planner.rviz file crashes and re-opening it imediately crashes rviz during landing phase
seems to have some relation with local planner publishing RViz-Goal topic (/goal_position) z position set to nan

**Log Files and Screenshots**
Only ROS side related problem,
Gazebo, PX4 SITL, QGC none of the programs gets affected, performs landing only rviz dies.

```
rviz: /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/include/OgreAxisAlignedBox.h:252: void Ogre::AxisAlignedBox::setExtents(const Ogre::Vector3&, const Ogre::Vector3&): Assertion `(min.x <= max.x && min.y <= max.y && min.z <= max.z) && "The minimum corner of the box must be less than or equal to maximum corner"' failed.
```

SITL Log
https://logs.px4.io/plot_app?log=2ba01be1-d4eb-4ef8-b718-1b600e2bf5c6

ROS bag file recorded from Hold -> Land mode
(double compressed for uploading)
[Local_Planner_SITL-Land-Rviz-kill.zip](https://github.com/PX4/avoidance/files/3767234/Local_Planner_SITL-Land-Rviz-kill.zip)

mrivi, 2019-10-24:
I cannot reproduce this issue.

SwiftGust, 2019-10-24:
Thank you for review.
Then this would be highly likely something is wrong with my desktop.
Is this could be an issue with ROS Melodic by chance?

SwiftGust, 2019-10-25:
After some digging, found my rviz dies when /path_waypoint topic (visualization_msgs::Marker) includes nan in the values.

closed 2019-10-25 by mrivi