PX4/PX4-Avoidance #521: Could not retrieve requested transform from buffer. Pointcloud dropped

germal, 2019-10-25:
Hi,

I am trying to start  local_planner_aero  on Jetson Nano with :

- Intel Realsense d435 camera
- Pixhawk FMUv2  with firmware 1.9.0 or 1.9.2  ( I tried the built one from source or from QGC)
- ROS Melodic 
- Mavros master -> 0.33.0
- Mavlink  2019.10.10

**roslaunch local_planner local_planner_aero.launch**

SUMMARY
========

PARAMETERS
 * /local_planner_node/goal_x_param: 0
 * /local_planner_node/goal_y_param: 0
 * /local_planner_node/goal_z_param: 4
 * /local_planner_node/pointcloud_topics: ['/camera/depth/c...
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /
    local_planner_node (local_planner/local_planner_node)
    tf_depth_camera (tf/static_transform_publisher)

ROS_MASTER_URI=localhost:11311

process[tf_depth_camera-1]: started with pid [14696]
process[local_planner_node-2]: started with pid [14697]
[ INFO] [1572023459.510298587]: Initializing nodelet with 4 worker threads.
[ WARN] [1572023459.876218953]: **Could not retrieve requested transform from buffer. Pointcloud dropped**
[ WARN] [1572023459.898790241]: Could not retrieve requested transform from buffer. Pointcloud dropped

 I have modified the "local_planner_aero.launch"  file in this way , changing the topic to reflect the color pointcloud of the d435 

arg name="pointcloud_topics" default="[/camera/depth/**color**/points]"

I rebuilt with many software version combinations and the result is always the same . Should be misconfiguration of mine , I use the following commands 

1) roslaunch mavros px4.launch fcu_url:=/dev/ttyUSB0:921600
2) roslaunch realsense2_camera rs_camera.launch filters:=pointcloud
3)roslaunch local_planner local_planner_aero.launch

Please help , I am struggling from weeks with this issue and I really don't know how to proceed with furher tests.
Thanks in advance !

germal, 2019-10-25:
These are the logs from Mavros  

 roslaunch mavros px4.launch fcu_url:=/dev/ttyUSB0:92160
.

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: /dev/ttyUSB0:921600
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/gps_uere: 1.0
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: base_link
 * /mavros/global_position/tf/frame_id: map
 * /mavros/global_position/tf/global_frame_id: earth
 * /mavros/global_position/tf/send: False
 * /mavros/global_position/use_relative_alt: True
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.0003490659 // 0...
 * /mavros/imu/frame_id: base_link
[... 191 lines cut ...]
[ INFO] [1572023219.209752525]: Built-in MAVLink package version: 2019.10.10
[ INFO] [1572023219.209825339]: Known MAVLink dialects: common ardupilotmega ASLUAV autoquad icarous matrixpilot paparazzi slugs standard uAvionix ualberta
[ INFO] [1572023219.209967268]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ INFO] [1572023219.214008225]: IMU: Attitude quaternion IMU detected!
[ INFO] [1572023219.218692267]: IMU: High resolution IMU detected!
[ INFO] [1572023219.513291339]: CON: Got HEARTBEAT, connected. FCU: PX4 Autopilot
[ INFO] [1572023219.518579036]: IMU: High resolution IMU detected!
[ INFO] [1572023219.519635566]: IMU: Attitude quaternion IMU detected!
[ INFO] [1572023220.535565636]: VER: 1.1: Capabilities         0x000000000000e4ef
[ INFO] [1572023220.536016686]: VER: 1.1: Flight software:     010902ff (106905871d000000)
[ INFO] [1572023220.536612321]: VER: 1.1: Middleware software: 010902ff (106905871d000000)
[ INFO] [1572023220.536998422]: VER: 1.1: OS software:         071c00ff (423371c7d4012e72)
[ INFO] [1572023220.537279677]: VER: 1.1: Board hardware:      00000011
[ INFO] [1572023220.537734268]: VER: 1.1: VID/PID:             26ac:0011
[ INFO] [1572023220.538215787]: VER: 1.1: UID:                 3335510a38343732
[ WARN] [1572023220.538984030]: CMD: Unexpected command 520, result 0
[ INFO] [1572023229.517245503]: HP: requesting home position
[ INFO] [1572023234.532071685]: WP: item #0* F:3 C: 22 p: 15.000000 0.000000 0.000000 nan x: 38.740643 y: -9.302164 z: 20.000000
[ INFO] [1572023234.546613294]: WP: item #1  F:3 C: 16 p: 0.000000 0.000000 0.000000 nan x: 38.740627 y: -9.302159 z: 20.000000
[ INFO] [1572023234.549816996]: WP: item #2  F:3 C: 16 p: 0.000000 0.000000 0.000000 nan x: 38.740559 y: -9.302008 z: 20.000000
[ INFO] [1572023234.555305426]: WP: item #3  F:2 C: 20 p: 0.000000 0.000000 0.000000 nan x: 0.000000 y: -317498284827322163791582879838896128.000000 z: 20.000000
[ INFO] [1572023234.556385497]: WP: mission received
[ INFO] [1572023239.516156424]: HP: requesting home position
[ INFO] [1572023249.515614954]: HP: requesting home position
[ INFO] [1572023259.514650575]: HP: requesting home position

JoeyLeeNPU, 2019-10-26:
I also meet the same problem, and my drone can not plan a path. So will your planner work better?

germal, 2019-10-26:
Actually it doesn't work .The error seems to me indicating that the pointcloud coming from the Realsense camera has a format not suitable for the planner. Please help to understand .
Thanks a lot,
Germal

jkflying, 2019-10-26:
You need PX4 Firmware 1.10 beta to use latest avoidance. This might fix your issue.

Also, I have a potential bugfix in #520 around transforming the point cloud, but this isn't your issue I think.

I'm not sure if local_planner_aero has been used in a long time...

germal, 2019-10-26:
Hi
I tried with 1.10 beta but I found the same error.
I think too that it is an issue regarding TF transformation but in my case it is not performance related , because  the planner  doesn't work.

> . I don't think local_planner_aero has been used in a long  time...  

Do you mean that avoidance doesn't work anymore on an hardware platform ?
Thanks
Germal

germal, 2019-11-08:
I think that this point It is worth clarifying. Do Avoidance work on an hardware platform ?
Please reply
Thanks
Germal

jkflying, 2019-11-08:
Yes, we fly the `local_planner` regularly using both Intel and ARM companion computers.

germal, 2019-11-08:
With a Realsense Depth camera ?

jkflying, 2019-11-08:
Yes, and also with Structure Core.

germal, 2019-11-08:
Could you please provide the software version combination and the steps that you have followed with Realsense ? 
I followed the readme and I have that issue on TF.
Thanks,
Germal

closed 2019-11-10 by germal