PX4/PX4-Avoidance #522: A target point far from the current position is set accidentally when planner running

JoeyLeeNPU, 2019-11-04:
When I set a target position using the local planner, a target point (a point between start and goal) far from the current position (red line in Rviz) will be set accidentally, then the drone rush to this point.  
Has anyone else has a similar problem? 

Thanks for help,
Joey

mrivi, 2019-11-04:
Could you please provide more information. Can you explain exactly what you're doing?

JoeyLeeNPU, 2019-11-04:
I set a target position far from my drone, then planner works well and my drone is flying to the target. But sometimes there exists a rush accidentally during the entire flight, I found that the (current control command is wrong) red line in rviz has a big jump. This problem happened twice, both when the drone was close to the target position.

mrivi, 2019-11-04:
can you share the flight control log?

closed 2019-11-13 by JoeyLeeNPU