PX4/PX4-Avoidance #523: TF Buffer: could not retrieve requested transform from buffer, unregistered

maerad7, 2019-11-11:
**Why am I getting an error?
The environment is drone upboard2. Please tell me why or how to find the error.**

avoidance version : 0.3.1
camera export : export CAMERA_CONFIGS="camera,realsense,819612070807,0.3,0.32,-0.11,0,0,0"

roslaunch local_planner avoidance.launch fcu_url:=/dev/ttyUSB4
... logging to /home/up2/.ros/log/6161c0d6-041f-11ea-9031-8ca982fea4f2/roslaunch-up2-UP-APL01-14467.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:34181/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 0
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: False
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 0
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 0
 * /camera/realsense2_camera/gyro_fps: 0
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
[... 274 lines cut ...]
[ INFO] [1573434709.866322455]: insert Infrared to Stereo Module
[ INFO] [1573434709.890631784]: SELECTED BASE:Depth, 0
[ INFO] [1573434709.920501420]: RealSense Node Is Up!
advertised as /camera/depth/image_rect_raw_drop
advertised as /camera/infra1/image_rect_raw_drop
[ERROR] [1573434710.440143992]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573434710.441506492]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573434710.442868227]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573434710.444088485]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573434710.445301250]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573434710.446605069]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573434710.447925021]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573434710.449150719]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573434710.450317182]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573434710.451507733]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573434710.452794238]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573434710.454021860]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573434710.455198446]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573434710.456357776]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573434710.457544633]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573434710.458800668]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573434710.459966169]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573434710.461164566]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573434710.462356707]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573434710.463512828]: TF Buffer: could not retrieve requested transform from buffer, unregistered

mrivi, 2019-11-11:
@maerad7 there was a recent fix for the transform buffers. Please try avoidance master.

maerad7, 2019-11-11:
**The error also occurs in the master version. Since px4_config.yaml is missing, I used version 0.31(px4_config.ymal<--version 0.31). What's the problem?**

roslaunch local_planner avoidance.launch fcu_url:=/dev/ttyUSB4
... logging to /home/up2/.ros/log/49ae8f1c-0462-11ea-badf-8ca982fea4f2/roslaunch-up2-UP-APL01-16840.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:43865/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 0
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: False
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 0
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 0
 * /camera/realsense2_camera/gyro_fps: 0
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
[... 272 lines cut ...]
[ INFO] [1573463083.142817716]: Plugin wind_estimation loaded
[ INFO] [1573463083.148770614]: Plugin wind_estimation initialized
[ INFO] [1573463083.148970937]: Autostarting mavlink via USB on PX4
[ INFO] [1573463083.149052158]: Built-in SIMD instructions: SSE, SSE2
[ INFO] [1573463083.149092724]: Built-in MAVLink package version: 2019.10.10
[ INFO] [1573463083.149151583]: Known MAVLink dialects: common ardupilotmega ASLUAV autoquad icarous matrixpilot paparazzi slugs standard uAvionix ualberta
[ INFO] [1573463083.149194821]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ INFO] [1573463083.155586683]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1573463083.156381145]: infra1 stream is enabled - width: 640, height: 480, fps: 15, Format: Y8
[ INFO] [1573463083.156519041]: setupPublishers...
[ INFO] [1573463083.169884982]: Expected frequency for depth = 30.00000
[ INFO] [1573463083.299252511]: Expected frequency for infra1 = 15.00000
[ INFO] [1573463083.410802182]: setupStreams...
[ INFO] [1573463083.412376015]: insert Depth to Stereo Module
[ INFO] [1573463083.412971476]: insert Infrared to Stereo Module
[ INFO] [1573463083.433337774]: SELECTED BASE:Depth, 0
[ INFO] [1573463083.461231263]: RealSense Node Is Up!
advertised as /camera/depth/image_rect_raw_drop
advertised as /camera/infra1/image_rect_raw_drop
[ERROR] [1573463083.714573472]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573463083.719729281]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573463083.725843323]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573463083.730166698]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573463083.735694000]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1573463083.740915025]

germal, 2019-11-11:
I have exactly the same issue : I updated with the latest master just yesterday and I used the generated avoidance.launch .Please find attached
[avoidance.txt](https://github.com/PX4/avoidance/files/3830481/avoidance.txt)
Please advice,
Thanks
Germal

germal, 2019-11-11:
I have no more that error after a clean rebuild 
 catkin_make clean avoidance

Jaeyoung-Lim, 2019-11-11:
@germal `catkin_make` has been deprecated. Please use `catkin build` using `python-catkin-tools`

germal, 2019-11-11:
Thanks Jaeyoung-Lim

maerad7, 2019-11-12:
**I'm still seeing errors in advertised as / camera / infra1 / image_rect_raw_drop, advertised as / camera / depth / image_rect_raw_drop.
I tried to build on master version as well, but I'm still having problems.**

maerad7, 2019-11-12:
[avoidance.txt](https://github.com/PX4/avoidance/files/3833856/avoidance.txt)
[image]
[image]

mrivi, 2019-11-12:
@maerad7 Please use the px4_config.yaml inside ~/catkin_ws/src/avoidance/avoidance/resource

If that file is not correct you won't get the correct transform tree hence the buffer errors. Once you have changed to use that px4_config.yaml you can see how the transforms tree is by running `rosrun tf view_frames`

maerad7, 2019-11-12:
**Adding baud fixed the error.**
PR: Unknown parameter to get: CP_DIST What is this message?

maerad7, 2019-11-13:
thanks mrivi
thanks Jaeyoung-Lim

closed 2019-11-13 by maerad7