PX4/PX4-Avoidance #524: off board make errors

Wayne-xixi, 2019-11-11:
Hello, I am new to px4 and trying to follow https://github.com/PX4/avoidance/tree/master, to catkin make the avoidance file. But i got this error below, anyone can help me?
`Errors     << offboard:make /home/fapsros/catkin_ws/logs/offboard/build.make.004.log
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/crt1.o: In function `_start':
(.text+0x20): undefined reference to `main'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/fapsros/catkin_ws/devel/.private/offboard/lib/offboard/offboard_node] Error 1
make[1]: *** [CMakeFiles/offboard_node.dir/all] Error 2
make: *** [all] Error 2
cd /home/fapsros/catkin_ws/build/offboard; catkin build --get-env offboard | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << offboard:make                          [ Exited with code 2 ]    
Failed    <<< offboard                               [ 12.3 seconds ]          
Abandoned <<< avoidance                              [ Unrelated job failed ]  
Abandoned <<< global_planner                         [ Unrelated job failed ]  
Abandoned <<< local_planner                          [ Unrelated job failed ]  
Abandoned <<< safe_landing_planner                   [ Unrelated job failed ]  
Abandoned <<< test_mavros                            [ Unrelated job failed ]  
Finished  <<< mavros_extras                          [ 59.5 seconds ]          
[build] Summary: 9 of 15 packages succeeded.                                   
[build]   Ignored:   1 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: 5 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 3 minutes and 23.2 seconds total.
`
Actually I need to make a gazebo simulation with avoidance.

Jaeyoung-Lim, 2019-11-11:
@Wayne-xixi  `catkin_make` has been deprecated. You need to use `catkin build` using `pyhon-catkin-tools `

Wayne-xixi, 2019-11-11:
> @Wayne-xixi `catkin_make` has been deprecated. You need to use `catkin build` using `pyhon-catkin-tools `

hello, thank you for reply! But I tried catkin build and got the same error. 
`catkin build -w ~/catkin_ws
------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/kinetic
`
`[build] Found '15' packages in 0.0 seconds.                                    
[build] Package table is up to date.                                           
Starting  >>> baxter_core_msgs                                                 
Starting  >>> baxter_maintenance_msgs                                          
Starting  >>> libmavconn                                                       
Starting  >>> mavros_msgs                                                      
Starting  >>> rethink_ee_description                                           
Finished  <<< rethink_ee_description                 [ 1.0 seconds ]           
Starting  >>> baxter_description                                               
Finished  <<< baxter_description                     [ 4.0 seconds ]           
Finished  <<< baxter_maintenance_msgs                [ 5.0 seconds ]           
Finished  <<< baxter_core_msgs                       [ 9.0 seconds ]           
Finished  <<< mavros_msgs                            [ 10.3 seconds ]          
Finished  <<< libmavconn                             [ 14.7 seconds ]          
Starting  >>> mavros                                                           
Finished  <<< mavros                                 [ 2 minutes and 41.0 seconds ]
Starting  >>> mavros_extras                                                    
Starting  >>> offboard                                                         
_______________________________________________________________________________
Errors     << offboard:make /home/fapsros/catkin_ws/logs/offboard/build.make.006.log
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/crt1.o: In function `_start':
(.text+0x20): undefined reference to `main'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/fapsros/catkin_ws/devel/.private/offboard/lib/offboard/offboard_node] Error 1
make[1]: *** [CMakeFiles/offboard_node.dir/all] Error 2
make: *** [all] Error 2
cd /home/fapsros/catkin_ws/build/offboard; catkin build --get-env offboard | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << offboard:make                          [ Exited with code 2 ]    
Failed    <<< offboard                               [ 7.1 seconds ]           
Abandoned <<< avoidance                              [ Unrelated job failed ]  
Abandoned <<< global_planner                         [ Unrelated job failed ]  
Abandoned <<< local_planner                          [ Unrelated job failed ]  
Abandoned <<< safe_landing_planner                   [ Unrelated job failed ]  
Abandoned <<< test_mavros                            [ Unrelated job failed ]  
^E[build - 03:48.0] [8/14 complete] [8/8 jobs] [0 queued] [1 failed] [mavros_...Finished  <<< mavros_extras                          [ 1 minute and 54.9 seconds ]
[build] Summary: 8 of 14 packages succeeded.                                   
[build]   Ignored:   1 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: 5 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 4 minutes and 51.8 seconds total. `

Jaeyoung-Lim, 2019-11-11:
@Wayne-xixi It seems like the build is failing due a package called offboard which is not part of this project

Wayne-xixi, 2019-11-11:
> @Wayne-xixi It seems like the build is failing due a package called offboard which is not part of this project

Thank you very much! solved!

closed 2019-11-11 by Wayne-xixi