PX4/PX4-Avoidance #525: wait_for_service(/local_planner_node/set_parameters): failed to contact, will keep trying

germal, 2019-11-11:
Avoidance : current Master
    Intel Realsense d435 camera
    Pixhawk FMUv2 with firmware 1.10
    ROS Melodic
    Mavros master -> 0.33.0
    Mavlink 2019.10.10

**The error triggers after
roslaunch local_planner avoidance.launch**

please find attached the avoidance.launch which I use
[avoidance.txt](https://github.com/PX4/avoidance/files/3832309/avoidance.txt)
please advice ,
Thanks

mluser@jetson:~$ roslaunch local_planner avoidance.launch 
... logging to /home/mluser/.ros/log/23374ea8-04a0-11ea-86ee-d6cee9c34fd1/roslaunch-jetson-6904.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server 192.168.1.1:34403/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 0
 * /camera/realsense2_camera/accel_frame_id: _accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: _accel_optical_frame
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: _aligned_depth_to...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: _aligned_depth_to...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: _aligned_depth_to...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: _aligned_depth_to...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: _aligned_depth_to...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: _aligned_depth_to...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: _link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: _color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: _color_optical_frame
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: _depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: _depth_optical_frame
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_pointcloud: True
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: _fisheye1_frame
 * /camera/realsense2_camera/fisheye1_optical_frame_id: _fisheye1_optical...
 * /camera/realsense2_camera/fisheye2_frame_id: _fisheye2_frame
 * /camera/realsense2_camera/fisheye2_optical_frame_id: _fisheye2_optical...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: _fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 0
[... 89 lines cut ...]
[ INFO] [1573489813.549236948]: color stream is enabled - width: 640, height: 480, fps: 15
[ INFO] [1573489813.549490274]: setupPublishers...
[ INFO] [1573489813.558930080]: Expected frequency for depth = 30.00000
[ INFO] [1573489813.664064732]: Expected frequency for color = 15.00000
[ INFO] [1573489813.718048758]: setupStreams...
[ INFO] [1573489813.722377163]: insert Depth to Stereo Module
[ INFO] [1573489813.723047298]: insert Color to RGB Camera
[ INFO] [1573489813.959271772]: Add Filter: pointcloud
[ INFO] [1573489813.961682321]: num_filters: 1
[ INFO] [1573489813.961842368]: SELECTED BASE:Depth, 0
[ INFO] [1573489813.964078599]: RealSense Node Is Up!
 11/11 11:30:14,010 WARNING [546786234752] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
 11/11 11:30:14,111 WARNING [546534584704] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
[ WARN] [1573489814.114193771]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
advertised as /camera/depth/image_rect_raw_drop
[WARN] [1573489814.656207]: wait_for_service(/local_planner_node/set_parameters): failed to contact, will keep trying
Waiting for service /local_planner_node/set_parameters...
[WARN] [1573489815.875606]: wait_for_service(/local_planner_node/set_parameters): failed to contact, will keep trying
[rqt_param_toggle-9] process has finished cleanly
log file: /home/mluser/.ros/log/23374ea8-04a0-11ea-86ee-d6cee9c34fd1/rqt_param_toggle-9*.log
[WARN] [1573489825.896404]: wait_for_service(/local_planner_node/set_parameters): failed to contact, will keep trying
[WARN] [1573489835.924538]: wait_for_service(/local_planner_node/set_parameters): failed to contact, will keep trying
[WARN] [1573489845.946596]: wait_for_service(/local_planner_node/set_parameters): failed to contact, will keep trying
[WARN] [1573489855.972493]: wait_for_service(/local_planner_node/set_parameters): failed to contact, will keep trying
[WARN] [1573489866.002758]: wait_for_service(/local_planner_node/set_parameters): failed to contact, will keep trying

mrivi, 2019-11-12:
it means that the parameters are under a different namespace and the service your are calling to set them in the launch file cannot find them. Once you have launch the planner you can run`rosparam list` to check under which namespace the parameter you are trying to change are.

germal, 2019-11-12:
Thanks @mrivi , I rebuilt my avoidance.launch with the generation script provided , correcting also the   px4_config.yaml path ,  as you adviced  at #523 

> Please use the px4_config.yaml inside ~/catkin_ws/src/avoidance/avoidance/resource > 

and now the local planner starts without issues
Thanks a lot !
Germal

closed 2019-11-12 by germal

kal1b, 2021-06-08:
> Thanks @mrivi , I rebuilt my avoidance.launch with the generation script provided , correcting also the px4_config.yaml path , as you adviced at #523
>
>
> and now the local planner starts without [...]

HOW it work?( What commands should be used to solve the problem? I just can't understand, I'm just getting used to it.