PX4/PX4-Avoidance #526: UML for PX4 Avoidance

germal, 2019-11-14:
Hello

Is it possible to check the PX4 Avoidance UML model ?
Thanks

germal, 2019-11-15:
@mrivi I mean if it could be available in order to understand more in deep the structure of the software.I'd like to study it.
Thanks,
germal

mrivi, 2019-11-15:
We don't have any. In the README you can find the master theses that the Local and Global Planners are based of. However there as been a lot of development since then so some algorithm steps described in the thesis are gone from the code and some implementations in the code are not described in the thesis.

germal, 2019-11-16:
Finally I've got the following UML diagrams from a software called Visual Paradigm.
I don't know how accurate they are because they have been generated automatically.
At first glance the reversing process is very easy (menu tools->code->instant reverse) but I think that could be optimized.
Anyway they could be used as rough reference .
[image]
[image]
[image]