PX4/PX4-Avoidance #527: FCU : no range data received, loitering

maerad7, 2019-11-15:
[image]

Print the message as shown above.
CP_DIST 6
CP_DELAY 0.2
CP_GO_NO_DATA 1
CP_GUIDE_ANG 30

Sometimes collision detection occurs and sometimes it doesn't.
There is one telemetry port, serial communication, and baud is 57600.

mrivi, 2019-11-15:
Could you please check at which rate the `/mavros/obstacle/send` topic is published by the planner? If the planner is running too slowly, the distance data rate to the Firmware is too little and you get into failsafe

maerad7, 2019-11-15:
average rate: 0.731
	min: 1.367s max: 1.367s std dev: 0.00000s window: 2
average rate: 0.718
	min: 1.367s max: 1.420s std dev: 0.02624s window: 3
no new messages
average rate: 0.729
	min: 1.326s max: 1.420s std dev: 0.03833s window: 4
average rate: 0.729
	min: 1.326s max: 1.420s std dev: 0.03319s window: 5
average rate: 0.730
	min: 1.326s max: 1.420s std dev: 0.02979s window: 6
no new messages
average rate: 0.729
	min: 1.326s max: 1.420s std dev: 0.02752s window: 7
average rate: 0.730
	min: 1.326s max: 1.420s std dev: 0.02557s window: 8
average rate: 0.729
	min: 1.326s max: 1.420s std dev: 0.02398s window: 9
no new messages
average rate: 0.729
	min: 1.326s max: 1.420s std dev: 0.02261s window: 10
average rate: 0.729
	min: 1.326s max: 1.420s std dev: 0.02146s window: 11
no new messages
average rate: 0.729
	min: 1.326s max: 1.420s std dev: 0.02050s window: 12
average rate: 0.729
	min: 1.326s max: 1.420s std dev: 0.01969s window: 13
average rate: 0.729
	min: 1.326s max: 1.420s std dev: 0.01920s window: 14
no new messages
average rate: 0.727
	min: 1.326s max: 1.422s std dev: 0.02254s window: 15
average rate: 0.729
	min: 1.326s max: 1.422s std dev: 0.02481s window: 16
average rate: 0.729
	min: 1.326s max: 1.422s std dev: 0.02403s window: 17
no new messages
average rate: 0.729
	min: 1.326s max: 1.422s std dev: 0.02339s window: 18
average rate: 0.729
	min: 1.326s max: 1.422s std dev: 0.02276s window: 19
no new messages
average rate: 0.727
	min: 1.326s max: 1.433s std dev: 0.02603s window: 20
average rate: 0.728
	min: 1.326s max: 1.433s std dev: 0.02758s window: 21
average rate: 0.729
	min: 1.326s max: 1.433s std dev: 0.02697s window: 22
no new messages
average rate: 0.729
	min: 1.326s max: 1.433s std dev: 0.02638s window: 23
average rate: 0.729
	min: 1.326s max: 1.433s std dev: 0.02580s window: 24
average rate: 0.729
	min: 1.326s max: 1.433s std dev: 0.02526s window: 25
no new messages
average rate: 0.729
	min: 1.326s max: 1.433s std dev: 0.02477s window: 26
average rate: 0.728
	min: 1.326s max: 1.433s std dev: 0.02592s window: 27
average rate: 0.729
	min: 1.323s max: 1.433s std dev: 0.02720s window: 28
no new messages
no new messages
average rate: 0.704
	min: 1.323s max: 2.737s std dev: 0.25460s window: 29
no new messages
average rate: 0.704
	min: 1.323s max: 2.737s std dev: 0.25018s window: 30
average rate: 0.705
	min: 1.323s max: 2.737s std dev: 0.24641s window: 31
no new messages
no new messages
average rate: 0.685
	min: 1.323s max: 2.748s std dev: 0.33762s window: 32
average rate: 0.686
	min: 1.323s max: 2.748s std dev: 0.33268s window: 33
no new messages
no new messages
average rate: 0.668
	min: 1.323s max: 2.748s std dev: 0.39470s window: 34
average rate: 0.670
	min: 1.323s max: 2.748s std dev: 0.38939s window: 35
no new messages
average rate: 0.671
	min: 1.323s max: 2.748s std dev: 0.38434s window: 36
modifyied baud = 921600

maerad7, 2019-11-15:
**Do I need to increase the value of cp_delay?**

mrivi, 2019-11-15:
The rate is way too low. If the range data are older than 500ms, the failsafe kicks in.
At which rate is the `/local_pointcloud` being published? it should be at least around 5-6 hz

maerad7, 2019-11-18:
**rostopic hz /local_pointcloud**
subscribed to [/local_pointcloud]
no new messages
no new messages
average rate: 0.730
	min: 1.371s max: 1.371s std dev: 0.00000s window: 2
no new messages
average rate: 0.730
	min: 1.370s max: 1.371s std dev: 0.00022s window: 3
average rate: 0.729
	min: 1.370s max: 1.377s std dev: 0.00297s window: 4
average rate: 0.729
	min: 1.368s max: 1.377s std dev: 0.00328s window: 5
no new messages
average rate: 0.729
	min: 1.368s max: 1.377s std dev: 0.00364s window: 6
average rate: 0.729
	min: 1.368s max: 1.377s std dev: 0.00332s window: 7
average rate: 0.729
	min: 1.366s max: 1.377s std dev: 0.00376s window: 8
no new messages
average rate: 0.726
	min: 1.366s max: 1.414s std dev: 0.01440s window: 9
average rate: 0.728
	min: 1.343s max: 1.414s std dev: 0.01723s window: 10
no new messages
average rate: 0.728
	min: 1.343s max: 1.414s std dev: 0.01635s window: 11
average rate: 0.728
	min: 1.343s max: 1.414s std dev: 0.01559s window: 12
average rate: 0.729
	min: 1.343s max: 1.414s std dev: 0.01503s window: 13
no new messages
average rate: 0.729
	min: 1.343s max: 1.414s std dev: 0.01444s window: 14
average rate: 0.728
	min: 1.343s max: 1.414s std dev: 0.01395s window: 15
average rate: 0.729
	min: 1.343s max: 1.414s std dev: 0.01357s window: 16
no new messages
average rate: 0.729
	min: 1.343s max: 1.414s std dev: 0.01318s window: 17
average rate: 0.729
	min: 1.343s max: 1.414s std dev: 0.01279s window: 18
no new messages
no new messages
average rate: 0.691
	min: 1.343s max: 2.741s std dev: 0.31369s window: 19
average rate: 0.691
	min: 1.343s max: 2.741s std dev: 0.30539s window: 20
no new messages
average rate: 0.694
	min: 1.343s max: 2.741s std dev: 0.29845s window: 21
average rate: 0.695
	min: 1.343s max: 2.741s std dev: 0.29173s window: 22
average rate: 0.697
	min: 1.343s max: 2.741s std dev: 0.28535s window: 23
no new messages
average rate: 0.698
	min: 1.343s max: 2.741s std dev: 0.27934s window: 24
average rate: 0.699
	min: 1.343s max: 2.741s std dev: 0.27373s window: 25
average rate: 0.701
	min: 1.343s max: 2.741s std dev: 0.26849s window: 26
no new messages
average rate: 0.702
	min: 1.343s max: 2.741s std dev: 0.26353s window: 27
average rate: 0.703
	min: 1.343s max: 2.741s std dev: 0.25883s window: 28
no new messages
average rate: 0.704
	min: 1.343s max: 2.741s std dev: 0.25433s window: 29
average rate: 0.705
	min: 1.343s max: 2.741s std dev: 0.25011s window: 30
average rate: 0.705
	min: 1.343s max: 2.741s std dev: 0.24593s window: 31
no new messages
average rate: 0.706
	min: 1.343s max: 2.741s std dev: 0.24216s window: 32
no new messages
average rate: 0.686
	min: 1.343s max: 2.741s std dev: 0.33035s window: 33
no new messages
average rate: 0.687
	min: 1.343s max: 2.741s std dev: 0.32563s window: 34
average rate: 0.688
	min: 1.343s max: 2.741s std dev: 0.32109s window: 35
no new messages
no new messages
average rate: 0.671
	min: 1.343s max: 2.741s std dev: 0.38236s window: 36
average rate: 0.673
	min: 1.343s max: 2.741s std dev: 0.37746s window: 37
no new messages
average rate: 0.674
	min: 1.343s max: 2.741s std dev: 0.37275s window: 38
average rate: 0.675
	min: 1.343s max: 2.741s std dev: 0.36827s window: 39
average rate: 0.677
	min: 1.343s max: 2.741s std dev: 0.36396s window: 40
no new messages
average rate: 0.678
	min: 1.343s max: 2.741s std dev: 0.35973s window: 41
no new messages
average rate: 0.664
	min: 1.343s max: 2.742s std dev: 0.40552s window: 42

darshanz, 2019-11-18:
I am also in the same situation. Any clues why the planner could be sending so slowly, could it be because of hardware limitations of the companion computer or can it be controlled at the software level?

mrivi, 2019-11-18:
for the simulation, it's most likely a hardware limitation of your laptop. 
On the vehicle what kind of companion are you using? What's the rate of the incoming pointcloud ?
What mavlink mode are you using?

darshanz, 2019-11-18:
Simulation worked okay, this problem was seen only while flying the vehicle. The companion computer is UP2 with  Intel® Atom ™ x5-E3940 having 8GB RAM. Realsense D435 attached via USB3 port is being used.
Incoming pointcloud rate is less than 1 it's between 0.5 to 0.8.
Not sure what exactly the mavlink mode is(I am new to mavlink). If you mean flight mode it was set to Position mode using RC.

mrivi, 2019-11-20:
You should increase the realsense pointcloud incoming rate to ~15Hz from the realsense launch file and then see the rate you get on `/local_pointcloud`
Furthermore be aware that USB3 causes huge interference with the GPS on the vehicle. We advise switching to USB2

maerad7, 2019-12-02:
It was solved by upgrading to tx2.

closed 2019-12-02 by maerad7

AldoCF96, 2020-05-27:
I have the same problem what is tx2 ?

maponarooo, 2022-05-02:
> You should increase the realsense pointcloud incoming rate to ~15Hz from the realsense launch file and then see the rate you get on `/local_pointcloud` Furthermore be aware that USB3 causes huge interference with the GPS on the vehicle. We advise switching to USB2

Hello~
Please let me know how to increase the realsense pointcloud incoming rate to ~15Hz.