PX4/PX4-Avoidance #535: undefined reference to `YAML::detail::memory::create_node()'

weiLiangHare, 2019-11-20:
my os is ubuntu 18.04 and ros is melodic
but build the avoidance node.error
and when 

error list:

/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `YAML::detail::memory::create_node()'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `vtable for YAML::InvalidNode'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `YAML::detail::node_data::empty_scalar[abi:cxx11]()'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `YAML::detail::node_data::begin()'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `vtable for YAML::BadSubscript'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `YAML::detail::node_data::set_null()'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `YAML::detail::node_data::end()'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `typeinfo for YAML::BadConversion'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `YAML::BadConversion::~BadConversion()'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `typeinfo for YAML::InvalidNode'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `typeinfo for YAML::BadSubscript'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `YAML::Load(std::istream&)'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `vtable for YAML::RepresentationException'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `vtable for YAML::Exception'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `vtable for YAML::BadConversion'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `YAML::InvalidNode::~InvalidNode()'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `YAML::BadSubscript::~BadSubscript()'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `YAML::detail::node_data::mark_defined()'
collect2: error: ld returned 1 exit status
avoidance/safe_landing_planner/CMakeFiles/waypoint_generator_node.dir/build.make:538: recipe for target '/home/bingo/project/px4_avoid/devel/lib/safe_landing_planner/waypoint_generator_node' failed
make[2]: *** [/home/bingo/project/px4_avoid/devel/lib/safe_landing_planner/waypoint_generator_node] Error 1
CMakeFiles/Makefile2:6385: recipe for target 'avoidance/safe_landing_planner/CMakeFiles/waypoint_generator_node.dir/all' failed
make[1]: *** [avoidance/safe_landing_planner/CMakeFiles/waypoint_generator_node.dir/all] Error 2
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `YAML::detail::memory::create_node()'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `vtable for YAML::InvalidNode'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `YAML::detail::node_data::empty_scalar[abi:cxx11]()'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `YAML::detail::node_data::begin()'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `vtable for YAML::BadSubscript'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `YAML::detail::node_data::set_null()'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `YAML::detail::node_data::end()'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `typeinfo for YAML::BadConversion'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `YAML::BadConversion::~BadConversion()'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `typeinfo for YAML::InvalidNode'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `typeinfo for YAML::BadSubscript'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `YAML::Load(std::istream&)'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `vtable for YAML::RepresentationException'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `vtable for YAML::Exception'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `vtable for YAML::BadConversion'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `YAML::InvalidNode::~InvalidNode()'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `YAML::BadSubscript::~BadSubscript()'
/home/bingo/project/px4_avoid/devel/lib/libavoidance.so: undefined reference to `YAML::detail::node_data::mark_defined()'

Jaeyoung-Lim, 2019-11-20:
@weiLiangHare I think you are missing libyaml-cpp-dev

weiLiangHare, 2019-11-21:
I have installed the libyaml-cpp-dev,but the errors still occur

is it the wrong version  ? 
 libyaml-cpp-dev                                 0.5.2-4ubuntu1                               amd64        YAML parser and emitter for C++ - development files

Jaeyoung-Lim, 2019-11-21:
@weiLiangHare It seems like the right version. Can you try adding [yaml_cpp_catkin](https://github.com/ethz-asl/yaml_cpp_catkin) to your work space?

weiLiangHare, 2019-11-22:
I have compiled successfully 。Still don't know why。Thank you for your reply.

and `dpkg -l |grep yaml` list 

ii  libyaml-0-2:amd64                            0.1.7-2ubuntu3                               amd64        Fast YAML 1.1 parser and emitter library
ii  libyaml-cpp-dev                              0.5.2-4ubuntu1                               amd64        YAML parser and emitter for C++ - development files
ii  libyaml-cpp0.5v5:amd64                       0.5.2-4ubuntu1                               amd64        YAML parser and emitter for C++
ii  libyaml-dev:amd64                            0.1.7-2ubuntu3                               amd64        Fast YAML 1.1 parser and emitter library (development)
ii  libyaml-libyaml-perl                         0.69+repack-1                                amd64        Perl interface to libyaml, a YAML implementation
ii  python-yaml                                  3.12-1build2                                 amd64        YAML parser and emitter for Python
ii  python3-yaml                                 3.12-1build2                                 amd64        YAML parser and emitter for Python3
bingo@ubuntu:~/project$

closed 2019-11-22 by weiLiangHare