PX4/PX4-Avoidance #536: Arming and taking off the drone

Cesaire-Njietche, 2019-11-24:
I have been able to set up the gazebo simulator correctly and the iris drone is well visible. The command `rosrun mavros mavsys mode -c OFFBOARD` shows that the mode has changed. The command `rosrun mavros mavsafety arm` arms the drone for a few seconds then disarms it without triggering the drone to take off.

Here is the ouput 

... logging to /home/katelynn/.ros/log/625fd22a-0eab-11ea-b0ef-902e1c94c569/roslaunch-katelynn-3921.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://katelynn:36697/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /local_planner_nodelet/goal_x_param: 17
 * /local_planner_nodelet/goal_y_param: 15
 * /local_planner_nodelet/goal_z_param: 3
 * /local_planner_nodelet/pointcloud_topics: ['/camera/depth/p...
 * /local_planner_nodelet/world_name: /home/katelynn/ca...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.0003490659 // 0...
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: local_origin
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: True
 * /mavros/local_position/tf/send_fcu: False
 * /mavros/mission/pull_after_gcs: True
[... 268 lines cut ...]

[ INFO] [1574601434.580792897, 10.256000000]: parameter MPC_ACC_UP_MAX is set from  nan to 4.000000 

[ INFO] [1574601434.581374998, 10.256000000]: parameter MPC_JERK_MAX is set from  nan to 20.000000 

[ INFO] [1574601434.581473260, 10.256000000]: parameter MPC_JERK_MIN is set from  nan to 10.000000 

[ INFO] [1574601434.581837698, 10.256000000]: parameter MPC_LAND_SPEED is set from  nan to 0.700000 

[ INFO] [1574601434.601560653, 10.276000000]: parameter MPC_TKO_SPEED is set from  nan to 1.500000 

[ INFO] [1574601434.601771647, 10.276000000]: parameter MPC_XY_CRUISE is set from  nan to 5.000000 

[ INFO] [1574601434.621628994, 10.296000000]: parameter MPC_Z_VEL_MAX_DN is set from  nan to 1.000000 

[ INFO] [1574601434.621696795, 10.296000000]: parameter MPC_Z_VEL_MAX_UP is set from  nan to 3.000000 

[ INFO] [1574601434.621880561, 10.296000000]: parameter NAV_ACC_RAD is set from  nan to 2.000000 

[ INFO] [1574601439.333097202, 15.000000000]: WP: mission received
[ WARN] [1574601533.613709513, 109.064000000]: CMD: Unexpected command 176, result 0
[ INFO] [1574601546.549794042, 121.944000000]: FCU: ARMED by Arm/Disarm component command
[ INFO] [1574601556.582249009, 131.964000000]: FCU: DISARMED by Auto disarm initiated
INFO  [logger] closed logfile, bytes written: 7835685
[ INFO] [1574601589.828776625, 165.168000000]: Setting goal: Frame:local_origin, Position(28.006, 11.956, 0.000), Orientation(0.000, 0.000, -0.667, 0.745) = Angle: -1.460

Jaeyoung-Lim, 2019-11-24:
@Cesaire-Njietche You need to be streaming setpoints to be able to enter offboard mode. Are you trying to run avoidance? which node are you running?

Cesaire-Njietche, 2019-11-24:
Trying to run avoidance. local_planner more precisely. global_planner is working correctly

mrivi, 2019-11-25:
@Cesaire-Njietche I can reproduce the issue. It seems that offboard takeoff is broken in local planner.
You have to way around this:
1. use the planner in mission modes
2. do an auto takeoff from QGC and switch into offboard mode afterwards

We'll try to fix the bug as soon as possible

Cesaire-Njietche, 2019-11-29:
Okay thanks a lot @mrivi for the insight.

sei-project, 2020-08-26:
We have the same problem. Any news about it?
We would like to use local planner in our project with ROS in Offboard Mode Avoidance without using QGC.
Thank you.

yxt916, 2020-09-11:
@sei-project 

I tested, not fixed, local planner still not working.

kimzheng1204, 2021-04-08:
Have you run local planner in Offboard Mode successfully? I can only do it with QGC.

pirate-lofy, 2021-04-30:
I think this can happen due preflight disarm, to overcome this bug u need to arm then switch to takeoff mode