PX4/PX4-Avoidance #537: HP: requesting home_position

QING-ML, 2019-11-26:
When I directly use model in folder avoidance. mavros requesting home_position. How do you set home_position in local planner.
Thanks

mrivi, 2019-11-27:
The home position is set by the flight controller. How are you launch the local planner?

QING-ML, 2019-11-27:
launch the local planner is right. I can get the home_position. But when i only use the models in package avoidance and mavros. There are no msgs from topic /mavros/home_position/home.
so I can't active the rotor and launch myself motion planner.

mrivi, 2019-11-27:
The models don't have any capabilities by themself. You need to launch the autopilot to do anything with them. See the documentation here https://dev.px4.io/v1.9.0/en/simulation/ros_interface.html#launching-rossimulation

closed 2019-11-27 by mrivi