PX4/PX4-Avoidance #538: TM : Time jump detected. Resetting time synchroniser

maerad7, 2019-11-29:
Not arming.

help me.

[ INFO] [1574997405.540789888]: ROS Node Namespace: camera
[ INFO] [1574997405.540842432]: Device Name: Intel RealSense D435
[ INFO] [1574997405.540879648]: Device Serial No: 825312070955
[ INFO] [1574997405.540912768]: Device FW version: 05.11.15.00
[ INFO] [1574997405.540951744]: Device Product ID: 0x0B07
[ INFO] [1574997405.540984352]: Enable PointCloud: Off
[ INFO] [1574997405.541014208]: Align Depth: Off
[ INFO] [1574997405.541043712]: Sync Mode: Off
[ INFO] [1574997405.541722816]: Device Sensors: 
[ INFO] [1574997405.542272320]: Stereo Module was found.
[ INFO] [1574997405.542414400]: RGB Camera was found.
[ INFO] [1574997405.542524896]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1574997405.542576768]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1574997405.542610496]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1574997405.542650848]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1574997405.542686912]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1574997405.542722624]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1574997405.543278752]: num_filters: 0
[ INFO] [1574997405.543329088]: Setting Dynamic reconfig parameters.
[ INFO] [1574997405.568222496]: Plugin setpoint_velocity initialized
[ INFO] [1574997405.568655520]: Plugin sys_status loaded
[ INFO] [1574997405.605556256]: Plugin sys_status initialized
[ INFO] [1574997405.605923168]: Plugin sys_time loaded
[ INFO] [1574997405.626567168]: TM: Timesync mode: MAVLINK
[ INFO] [1574997405.631787904]: Plugin sys_time initialized
[ INFO] [1574997405.632254080]: Plugin trajectory loaded
[ INFO] [1574997405.688458528]: Plugin trajectory initialized
[ INFO] [1574997405.688884640]: Plugin vfr_hud loaded
[ INFO] [1574997405.691370880]: Plugin vfr_hud initialized
[ INFO] [1574997405.691481248]: Plugin vibration blacklisted
[ INFO] [1574997405.691761536]: Plugin vision_pose_estimate loaded
[ INFO] [1574997405.755335680]: Plugin vision_pose_estimate initialized
[ INFO] [1574997405.755882496]: Plugin vision_speed_estimate loaded
[ INFO] [1574997405.804118336]: Plugin vision_speed_estimate initialized
[ INFO] [1574997405.804876544]: Plugin waypoint loaded
[ INFO] [1574997405.867279584]: Plugin waypoint initialized
[ INFO] [1574997405.867478144]: Plugin wheel_odometry blacklisted
[ INFO] [1574997405.868134272]: Plugin wind_estimation loaded
[ INFO] [1574997405.875830272]: Plugin wind_estimation initialized
[ INFO] [1574997405.876114464]: Autostarting mavlink via USB on PX4
[ INFO] [1574997405.876322592]: Built-in SIMD instructions: ARM NEON
[ INFO] [1574997405.876447456]: Built-in MAVLink package version: 2019.11.11
[ INFO] [1574997405.876581600]: Known MAVLink dialects: common ardupilotmega ASLUAV autoquad icarous matrixpilot paparazzi slugs standard uAvionix ualberta
[ INFO] [1574997405.876699296]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ INFO] [1574997405.884652416]: CON: Got HEARTBEAT, connected. FCU: PX4 Autopilot
[ERROR] [1574997405.892702368]: FCU: REJECT POSITION CONTROL
[ INFO] [1574997405.893391360]: IMU: Attitude quaternion IMU detected!
[ INFO] [1574997405.893660480]: IMU: High resolution IMU detected!
[ INFO] [1574997405.894097184]: RC_CHANNELS message detected!
[ INFO] [1574997405.938179904]: Done Setting Dynamic reconfig parameters.
[ERROR] [1574997405.942341152]: FCU: REJECT POSITION CONTROL
[ERROR] [1574997405.988139488]: FCU: No range data received, no movement allowed.
[ERROR] [1574997406.036890464]: FCU: No range data received, no movement allowed.
[ERROR] [1574997406.085796000]: FCU: No range data received, no movement allowed.
[ INFO] [1574997406.893116928]: VER: 1.1: Capabilities         0x000000000000e4ef
[ INFO] [1574997406.893254912]: VER: 1.1: Flight software:     010a0080 (4964b633b2000000)
[ INFO] [1574997406.893328064]: VER: 1.1: Middleware software: 010a0080 (4964b633b2000000)
[ INFO] [1574997406.893387488]: VER: 1.1: OS software:         00000000 (66401c8960c5211d)
[ INFO] [1574997406.893438368]: VER: 1.1: Board hardware:      00000011
[ INFO] [1574997406.893484256]: VER: 1.1: VID/PID:             26ac:0011
[ INFO] [1574997406.893528160]: VER: 1.1: UID:                 3337510d37353832
[ WARN] [1574997406.895116928]: CMD: Unexpected command 520, result 0
[ INFO] [1574997406.926283456]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1574997406.927018656]: infra1 stream is enabled - width: 640, height: 480, fps: 15
[ INFO] [1574997406.927158048]: setupPublishers...
[ INFO] [1574997406.936333440]: Expected frequency for depth = 30.00000
[... 83 lines cut ...]
[ INFO] [1574997417.903308128]: parameter MPC_ACC_HOR is set from  nan to 3.000000 

[ INFO] [1574997417.906726400]: parameter MPC_ACC_UP_MAX is set from  nan to 4.000000 

[ INFO] [1574997417.916186400]: parameter MPC_AUTO_MODE is set from -1 to 1 

[ INFO] [1574997417.948237888]: parameter MPC_JERK_MAX is set from  nan to 20.000000 

[ INFO] [1574997417.978534912]: parameter MPC_LAND_SPEED is set from  nan to 0.700000 

[ INFO] [1574997418.059428000]: parameter MPC_TKO_SPEED is set from  nan to 1.500000 

[ INFO] [1574997418.076909376]: parameter MPC_XY_CRUISE is set from  nan to 5.000000 

[ INFO] [1574997418.170043008]: parameter MPC_Z_VEL_MAX_UP is set from  nan to 3.000000 

[ INFO] [1574997418.179223136]: parameter NAV_ACC_RAD is set from  nan to 2.000000 

[ INFO] [1574997420.896251840]: WP: mission received
 29/11 12:20:39,591 ERROR [547583160720] (ds5-options.cpp:82) Asic Temperature value is not valid!
[ WARN] [1574997639.856628544]: GP: No GPS fix
[ERROR] [1574997657.437591255]: TM : Time jump detected. Resetting time synchroniser.
[ WARN] [1574997670.626892193]: GP: No GPS fix
[ WARN] [1574997700.638812984]: GP: No GPS fix
[ERROR] [1574997711.437008914]: TM : Time jump detected. Resetting time synchroniser.

baumanta, 2019-11-29:
This is not very helpful. Please provide more information about where you are running and what is happening. Is this SITL or on a board? If so, which board. Also I see: [ INFO] [1574997405.540984352]: Enable PointCloud: Off
So does the camera publish a pointcloud? If it is not arming, what message do you get?

maerad7, 2019-11-29:
Sorry
Information 
jetson Tx2 + j120 board
Intel d435 camera
os ubuntu 16.04
px4 avoidance version -> master
realsense sdk version( jetsonhacks install version )v2.13.0
reasense-ros wrapper 2.2.8

OK point cloud off....

I will solve the problem, but I hope you can let me know if there is a way.

baumanta, 2019-11-29:
you probably need to change some parameter in the realsense launch file. You can verify if the pointcloud is really published by `rostopic hz /pointcloud_topic_name`

maerad7, 2019-11-29:
Thank you very much. I'll try it in two days.

maerad7, 2019-12-02:
I solved it by restarting.
thanks

closed 2019-12-02 by maerad7

IngTIKNA, 2023-05-27:
The same error occurs when PX4(already armed) is connected to the QGroundControl

BEI11HAI, 2023-08-02:
@IngTIKNA Did you solve the issue? I'm in the same situation as well