PX4/PX4-Avoidance #539: Failed to start local planner, Local planner hangs at "Planner abort: missing required data ".

shrazshad, 2019-12-03:
cribe the bug**
1. Local planner hangs at "Planner abort: missing required data ".
2. [image]
3. rostopic list | grep points shows nothing related to depth camera, i think depth camera somehow failed to be turned on?

**To Reproduce**
1. using either px4 1.9.2 or master has the same result.
2. using : roslaunch local_planner local_planner_depth-camera.launch

**Expected behavior**
Planner should be ready to be switched to offboard mode and fly

**Log Files and Screenshots**
*Always* provide a link to the flight log file:

`SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /local_planner_nodelet/goal_x_param: 17
 * /local_planner_nodelet/goal_y_param: 15
 * /local_planner_nodelet/goal_z_param: 3
 * /local_planner_nodelet/pointcloud_topics: ['/camera/depth/p...
 * /local_planner_nodelet/world_name: /home/shr/softwar...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.0003490659 // 0...
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: local_origin
 * /mavros/local_position/tf/child_frame_id: fcu
[... 274 lines cut ...]
WARN  [commander] Avoidance system rejected
[ERROR] [1575346809.935751947, 15.216000000]: FCU: Avoidance system rejected
since_last_cloud: 15.284
[ WARN] [1575346810.004116297, 15.284000000]:  Planner abort: missing required data 
 
since_last_cloud: 15.384
[ WARN] [1575346810.104153280, 15.384000000]:  Planner abort: missing required data 
 
since_last_cloud: 15.484
[ WARN] [1575346810.203717628, 15.484000000]:  Planner abort: missing required data 
 
since_last_cloud: 15.584
[ WARN] [1575346810.304611837, 15.584000000]:  Planner abort: missing required data 
 
since_last_cloud: 15.684
[ WARN] [1575346810.403636776, 15.684000000]:  Planner abort: missing required data 
 
since_last_cloud: 15.784
[ WARN] [1575346810.504258717, 15.784000000]:  Planner abort: missing required data 
 
^Csince_last_cloud: 15.884
[ WARN] [1575346810.603954975, 15.884000000]:  Planner abort: missing required data`

**Additional context**
Add any other context about the problem here.

mrivi, 2019-12-03:
Can you see the vehicle and the world visualization in rviz? Could you please share the output of `echo $GAZEBO_MODEL_PATH` and `echo $GAZEBO_PLUGIN_PATH` here?

khbnzr, 2020-01-23:
@mrivi My issue is similar. The output of - 
echo $GAZEBO_PLUGIN_PATH - /usr/lib/x86_64-linux-gnu/gazebo-7/plugins:
echo $GAZEBO_MODEL_PATH - have both /avoidance/avoidance/sim/worlds & /avoidance/avoidance/sim/models.

What could be the possible reason for the above logs by @shrazshad. Thanks in advance.

baumanta, 2020-01-23:
The planner basically needs to things in order to work:
- information from the flight controller (/mavros/local_position/pose, ...)
- pointcloud

Try to do rostopic list and then look at the rates at which you are getting this information. If either of them is not published or way too slow you will see the mentioned warnings. If the pointcloud is missing, you might have problems with the gazebo plugin or model of the camera/drone. If the stuff from mavros is missing, you might have wrong fcu url or did not build the firmware correctly.

khbnzr, 2020-01-24:
@baumanta I am getting the data on /mavros/local_position/pose topic but not on pointcloud. I've installed the gazebo using apt ros-kinetic-gazebo-* , which installs gazebo-7. I'm using this drone/camera model from the launch file - https://github.com/PX4/avoidance/blob/master/avoidance/launch/avoidance_sitl_mavros.launch#L41-L44. I've uninstalled and re-installed the gazebo-7 but the issue was same. Am i missing something here ?

PonyZhou, 2020-03-30:
I had the same issue when launching local_planner_realsense.launch. And i found the pointclod is not published. Since the realsense camera can publish /depth/points topic directly, i wonder whether PX4/disparity_to_point_cloud is necessary.

nicovanduijn, 2020-03-30:
Note that the `local_planner_realsense.launch` is a launchfile to run using actual hardware, while @khbnzr seems to be working in simulation.

@PonyZhou We experimented with both the pointcloud generated by the device directly, as well as the point cloud generation based on disparity a while ago, and decided to stick with this option. I believe there were some performance differences, but I don't exactly recall. Feel free to try the both, I'm sure we'd love to hear a conclusive investigation on their differences in terms of CPU load & accuracy

PonyZhou, 2020-03-30:
Thanks for reply! And i did not make the code of PX4/disparity_to_point_cloud, but when i launched local_planner_realsense.launch,  the same waring "Planner abort: missing required data " occurred. So I guess local_planner node cannot use /depth/points from realsense by default.

PonyZhou, 2020-03-31:
How can local_planner node use /depth/points from realsense directly without disparity_to_point_cloud?

nicovanduijn, 2020-03-31:
It was a long time ago I tried this, and things can change quickly, you will have to do some digging on your own there. I vaguely recall that I started here: https://github.com/PX4/avoidance/blob/c0c159bb6d260e323a9140c8f277daf6db2bdecb/local_planner/launch/rs_depthcloud.launch#L31
and then you have to adjust the the transform trees correctly, which I don't recall the details of. There may be more idiosyncrasies of the realsense that make your life miserable when trying to do this, such as bandwidth limitations on USB2.0 (some streams missing, aka. you can't have a pointcloud and RGB image concurrently or things like that).

This is not a support forum, so please refrain from opening issues just to ask for help. There's the forum and slack channels for this.

closed 2020-03-31 by nicovanduijn

PonyZhou, 2020-03-31:
Thanks for your instruction and suggestion.