PX4/PX4-Avoidance #542: [Question] Can I run local_planner in 2D mode?

heleidsn, 2019-12-10:
Hello!

I am going do to some RL-based navigation and I want to simplify the problem from 3D to 2D because leaning a 3D planner is quite difficult. **Can I run local_planner in 2D mode, which means avoid obstacle without altitude change?** 

I have tied to change `pitch_cost_param_` up to 1000, but there is still a change in height. Does anyone have some ideas about this? If there is some way to do this with the parameter settings or with small code changes? 

Thank you!

Jaeyoung-Lim, 2020-01-28:
@heleidsn If you are trying to use the planner as a expert for learning, you can process your data so that it doesn't take altitude as the data. In general, to effectively generate the dataset, it is better to not run the planner as in flight, but feed in data(pointlcoud) and generate paths directly out of the planner function.

heleidsn, 2020-02-23:
@Jaeyoung-Lim Thanks for your advice! I just need a little bit of data and the work has finished. I will close this issue.

closed 2020-02-23 by heleidsn