PX4/PX4-Avoidance #543: NANs in LAND mode

baumanta, 2019-12-11:
When the autopilot is in LAND mode (and maybe some others) it sends a goal=[x, y, NAN]. On the local planner side this goal through the whole calculation resulting in a cost-matrix full of NANs etc. This does not really affect the performance, as we are not doing any avoidance maneuvers in land/takeoff mode and the planner recovers as soon as it gets a valid goal again. But still this should be fixed, also because processing NANs is slow and should be avoided anyway.

Test: If you display the /cost_matrix topic in RVIZ,  the matrix will turn yellow if it contains NANs.

mrivi, 2019-12-12:
This is not a bug, meaning that NAN in PX4 is a not controlled output. In the case of a land, PX4 doesn't control the z position but only velocity. If we set the position we will never be able to land properly given the error in z. 
The only thing we can do is set the goal position in z to zero but then we need to ignore that position on the waypoint generation side.

baumanta, 2019-12-12:
yes, that is what I mean. I think it needs some discussion on how to best handle it, but processing all NANs through the local_planner is not a good solution in my opinion. It is not urgent, I just wanted to note it here.