PX4/PX4-Avoidance #545: CMake Error

Sunchanghao, 2019-12-18:
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
YAML_CPP_INCLUDE_DIR
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/global_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner
   used as include directory in directory /home/sch/catkin_ws/src/avoidance/local_planner

Jaeyoung-Lim, 2019-12-18:
@Sunchanghao Can you check if yaml_cpp is insalled? try
`sudo apt install ros-kinetic-yaml-*`
if you are running ros kinetic

Sunchanghao, 2019-12-19:
@Jaeyoung-Lim Thx I have installed this:ros-kinetic-yaml-cpp-0-3
when my fist time to install this node,it has no questiion
after I use another computer download this node and build it,cmake error

mrivi, 2020-01-06:
plaese make sure you have run `sudo apt install ros-kinetic-yaml-*` on all your machines.

closed 2020-01-06 by mrivi