PX4/PX4-Avoidance #546: where is safety margin variable?

maerad7, 2019-12-24:
We use GitHub issues only to discuss PX4 Avoidance bugs and new features. For
questions about using PX4 or related components, please use [PX4 Discuss](http://discuss.px4.io/).

Thanks!

nicovanduijn, 2019-12-24:
[This] (https://github.com/PX4/avoidance/blob/af5b0351317cd680b11696399d2ca32f42e70a5a/local_planner/cfg/LocalPlannerNode.cfg#L15) parameter is closest to what you're looking for I think. But note that it does not guarantee that minimum distance at all times

closed 2019-12-24 by nicovanduijn

maerad7, 2019-12-24:
Does it mean 8.5, 0.0, 30?
Thank you for your prayer.

nicovanduijn, 2019-12-24:
It's a parameter you can change dynamically through RQT reconfigure, I suggest doing so rather than changing that line. The 8.5/0/30 are default/min/max respectively