PX4/PX4-Avoidance #547: Px4 Avoidance Local Planner Testing on Hardware

harishmore05, 2019-12-26:
Hii,
I am facing tf error while excuating avoidance.launch file.
I have generated avoidance.launch file as specified in Px4/Avoidance github page.

and **avoidance.launch** file is as like:

<launch>
    <arg name="ns" default="/"/>
    <arg name="fcu_url" default="/dev/ttyACM0:57600"/>
    <arg name="gcs_url" default="" />   
    <arg name="tgt_system" default="1" />
    <arg name="tgt_component" default="1" />

    
    <group ns="$(arg ns)">
        <include file="$(find mavros)/launch/node.launch">
            <arg name="pluginlists_yaml" value="$(find mavros)/launch/px4_pluginlists.yaml" />
            
            <arg name="config_yaml" value="$(find local_planner)/resource/px4_config.yaml" />
            <arg name="fcu_url" value="$(arg fcu_url)" />
            <arg name="gcs_url" value="$(arg gcs_url)" />
            <arg name="tgt_system" value="$(arg tgt_system)" />
            <arg name="tgt_component" value="$(arg tgt_component)" />
        </include>
    </group>

    
<node pkg="tf" type="static_transform_publisher" name="tf_camera" required="true"
 args="0.3 0.32 -0.11 0 0 0 fcu camera_link 100"/> 

<include file="$(find local_planner)/launch/rs_depthcloud.launch">
<arg name="required"              value="true"/>
<arg name="namespace"             value="camera" />
<arg name="tf_prefix"             value="camera" />
<arg name="serial_no"             value="828112071698"/>
<arg name="depth_fps"             value="30"/>
</include>

      
      <node name="drop_camera_depth" pkg="topic_tools" type="drop" output="screen"
        args="/camera/depth/image_rect_raw 29 30">
      </node>
      <node name="drop_camera_ir" pkg="topic_tools" type="drop" output="screen"
        args="/camera/infra1/image_rect_raw 29 30">
      </node>
    <node name="dynparam" pkg="dynamic_reconfigure" type="dynparam" args="load local_planner_node /home/harish/catkin_ws/src/avoidance/local_planner/cfg/vehicle.yaml" />
    
    <env name="ROSCONSOLE_CONFIG_FILE" value="$(find local_planner)/resource/custom_rosconsole.conf"/>
    <arg name="pointcloud_topics" default="[/camera/depth/points]"/>

    <node name="local_planner_node" pkg="local_planner" type="local_planner_node" output="screen" required="true" >
      <param name="goal_x_param" value="0" />
      <param name="goal_y_param" value="0"/>
      <param name="goal_z_param" value="4" />
      <rosparam param="pointcloud_topics" subst_value="True">$(arg pointcloud_topics)</rosparam>
    </node>
    
    
    <node name="rqt_param_toggle" pkg="local_planner" type="rqt_param_toggle.sh" />

</launch>

**px4_config.yaml:**

# Common configuration for PX4 autopilot
#
# node:
startup_px4_usb_quirk: true

# --- system plugins ---
[... 681 lines cut ...]
[ERROR] [1577346925.783024708]: ODOM: Ex: Could not find a connection between 'fcu' and 'map_ned' because they are not part of the same tree.Tf has two or more unconnected trees.
advertised as /camera/infra1/image_rect_raw_drop
advertised as /camera/depth/image_rect_raw_drop
[ INFO] [1577346926.663375544]: VER: 1.1: Capabilities         0x000000000000e4ef
[ INFO] [1577346926.663466182]: VER: 1.1: Flight software:     010902ff (106905871d000000)
[ INFO] [1577346926.663525961]: VER: 1.1: Middleware software: 010902ff (106905871d000000)
[ INFO] [1577346926.663575567]: VER: 1.1: OS software:         071c00ff (423371c7d4012e72)
[ INFO] [1577346926.663621916]: VER: 1.1: Board hardware:      00000011
[ INFO] [1577346926.663660062]: VER: 1.1: VID/PID:             26ac:0011
[ INFO] [1577346926.663696865]: VER: 1.1: UID:                 3036510b31353833
[ WARN] [1577346926.667518655]: CMD: Unexpected command 520, result 0
[ERROR] [1577346926.805901849]: ODOM: Ex: Could not find a connection between 'fcu' and 'map_ned' because they are not part of the same tree.Tf has two or more unconnected trees.
[rqt_param_toggle-11] process has finished cleanly
log file: /home/harish/.ros/log/1242e822-27b5-11ea-8d5b-98fa9b867e40/rqt_param_toggle-11*.log
[ERROR] [1577346927.566279760]: TF Buffer: could not retrieve requested transform from buffer, unregistered
[ERROR] [1577346927.571395632]: TF Buffer: could not retrieve requested transform from buffer, unregistered

**Rqt_graph:**
[image]

**and tf_view_frames:**
[frames.pdf](https://github.com/PX4/avoidance/files/4001832/frames.pdf)

I think I am facing errors of tf frames mismatched.
Help me get rid of this errors.

nicovanduijn, 2020-01-06:
Please take the time to post your question in a readable and understandable fashion. 

I cannot follow your post. If you want other people to invest time into helping you debug your problem, it helps a lot if you ask concisely and in a easy-to-read post.

If you put the px4 config in a comment block, it should get rid of the awkward formatting. Same for the  console output. You can edit your post and we'll try to understand your issue

harishmore05, 2020-01-06:
Hi @nicovanduijn,
Sorry for awkward formatting, now I updated my issue.

baumanta, 2020-01-08:
@harishmore05 I think your problem is the px4 config. Look at the one in this repo for guidance. e.g your `map` frame should be called `local_origin`

AhmedElsafy, 2020-01-22:
@baumanta Is this the only way to run the package? I have changed map to local_origin but I had error with other packages?

MVTERRY, 2020-05-02:
@baumanta  If you look in this repo the px4 config file has `map` as its `odom_parent_id_des` frame. I am having the same problem as the author here. When I changed the frame to local_origin I got a new error.
  `ODOM: Ex: "map_ned" passed to lookupTransform argument source_frame does not exist. 
`

Cristian-wp, 2020-07-16:
Hello, I have a similar problem reported in my issue #594 

@harishmore05 have you find a solution? If yes can you explane how do you resolve this problem?
@baumanta can you please take a look at my issue? I need help and I do not what to do to fix it :(