PX4/PX4-Avoidance #550: [global_planner_node-12] process has died

arnozeelmaekers, 2020-01-08:
PX4 Firmware version: v1.10.0
Mavros version: 0.33.3-1
Ubuntu: 18.04
Avoidance: Master branch

Note: Everything is running inside of a Docker container on a Ubuntu 18.04 machine.

Bug: when trying to run Global planner the default value for octomap_server resolution is 1 and for global_planner_node robot_radius is 0.5 however due to the project I'm working on (Indoor flight) I need a higher resolution. What I've noticed is that if I put the resolution below the robot_radius value after a while it throws:
```
[global_planner_node-12] process has died [pid 26643, exit code -11, cmd /home/user/Projects/catkin_ws/devel/lib/global_planner/global_planner_node /home/user/Projects/catkin_ws/src/avoidance/global_planner/resource/random_goals __name:=global_planner_node __log:=/home/user/.ros/log/d0910c1e-31e7-11ea-abeb-0242ac110002/global_planner_node-12.log].
log file: /home/user/.ros/log/d0910c1e-31e7-11ea-abeb-0242ac110002/global_planner_node-12*.log
```
Note: The log file is nowhere to be found.
If I lower the robot_radius below the resolution it no longer gives the error or at least from my testing. But lowering the robot_radius is not an option since it causes the drone to hit things.

Any suggestion on what this could be?

mrivi, 2020-01-13:
@arnozeelmaekers - try to reproduce the failure with valgrind running. You can see [here](http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB) how to do it. Also build the global planner is debug mode `catkin config --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCATKIN_ENABLE_TESTING=True && catkin build global_planner` before running it with valgrind.  The simulation will run super slowly but it should point you directly to the line causing the seg fault

Jaeyoung-Lim, 2020-01-23:
@arnozeelmaekers I also see this without decreasing the resolution