PX4/PX4-Avoidance #552: [Global Planner] Unable to cope with RTL & LAND mode

SwiftGust, 2020-01-21:
**Describe the bug**
A clear and concise description of what the bug is.
Global planner on the master is not able to handle with RTL & Land mode since /mavros/trajectory/desired topic comes in does not contain valid data, no frame_id
while the local planner could work with RTL & Land mode from the firmware.
```
$ rostopic echo mavros/trajectory/desired

---
header: 
  seq: 3574
  stamp: 
    secs: 721
    nsecs: 392112295
  frame_id: "local_origin"
type: 0
point_1: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  coordinate_frame: 0
  type_mask: 0
  position: 
    x: -0.0537663251162
    y: 0.0184620078653
    z: 29.9580383301
  velocity: 
    x: nan
    y: nan
    z: nan
  acceleration_or_force: 
    x: nan
    y: nan
    z: nan
  yaw: nan
  yaw_rate: nan
point_2: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  coordinate_frame: 0
  type_mask: 0
  position: 
    x: -0.0537663251162
    y: 0.0184620078653
    z: 29.9580383301
  velocity: 
    x: nan
    y: nan
    z: nan
  acceleration_or_force: 
    x: nan
    y: nan
    z: nan
  yaw: 2.80641579628
  yaw_rate: nan
point_3: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  coordinate_frame: 0
  type_mask: 0
  position: 
    x: -0.0537663251162
    y: 0.0184620078653
    z: 29.9580383301
  velocity: 
    x: nan
    y: nan
    z: nan
  acceleration_or_force: 
    x: nan
    y: nan
    z: nan
  yaw: nan
  yaw_rate: nan
point_4: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  coordinate_frame: 0
  type_mask: 0
  position: 
    x: nan
    y: nan
    z: nan
  velocity: 
    x: nan
    y: nan
    z: nan
  acceleration_or_force: 
    x: nan
    y: nan
    z: nan
  yaw: nan
  yaw_rate: nan
point_5: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  coordinate_frame: 0
  type_mask: 0
  position: 
    x: nan
    y: nan
    z: nan
  velocity: 
    x: nan
    y: nan
    z: nan
  acceleration_or_force: 
    x: nan
    y: nan
    z: nan
  yaw: nan
  yaw_rate: nan
point_valid: [1, 1, 1, 0, 0]
command: [65535, 65535, 65535, 65535, 65535]
time_horizon: [0.0, 0.0, 0.0, 0.0, 0.0]
---
```
**To Reproduce**
Steps to reproduce the behavior:
1. Launch SITL
2. Uploaded mission any mission ends with RTL or LAND
[mission_ends_RTL.zip](https://github.com/PX4/avoidance/files/4090041/mission_ends_RTL.zip)
3. Start mission 
4. See what happens when the vehicle enters RTL or Land Mode

**Expected behavior**
A clear and concise description of what you expected to happen.
In RTL or Land mode, the vehicle hover at current position and does nothing.

Jaeyoung-Lim, 2020-01-21:
@SwiftGust Wouldn't the correct expected behavior be the vehicle RTLs or Land? Why do you want it do hover at the current position?

SwiftGust, 2020-01-21:
That's miss-stated, edited.

Jaeyoung-Lim, 2020-01-21:
@SwiftGust This is related to how the setpoints are handled in global planner. This was partly the reason I was trying to port the state machine into global_planner in https://github.com/PX4/avoidance/pull/506, but this is still unfinished

SwiftGust, 2020-01-21:
For RTL and Land, it seems its waypoint is given well however, in the callbackhttps://github.com/PX4/avoidance/blob/master/global_planner/src/nodes/global_planner_node.cpp#L247

It doesn't seem it is blocked to use but why the global planner is not able to cope it?

Jaeyoung-Lim, 2020-01-21:
@SwiftGust Do you have a log? Where does it stop following the land point?

SwiftGust, 2020-01-21:
@Jaeyoung-Lim When RTL is enabled, It seems path finder behind is unable to find a path with altitude difference (it is lower than planner's max_altitude_ though)

my case, mission is at 10m -> RTL 30m, reducing rtl_return_alt to 0 enabled return but still unable to land, it's z is nan for point_1 but in point_2, it is valid point.

Log : https://logs.px4.io/plot_app?log=0119da93-0352-4c0f-b734-80abb2704388

topic /mavros/trajectory/desired
```
---
header: 
  seq: 4050
  stamp: 
    secs: 823
    nsecs: 320140337
  frame_id: "local_origin"
type: 0
point_1: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  coordinate_frame: 0
  type_mask: 0
  position: 
    x: 0.0291219893843
    y: -0.0139899933711
    z: nan
  velocity: 
    x: nan
    y: nan
    z: -1.0
  acceleration_or_force: 
    x: nan
    y: nan
    z: nan
  yaw: nan
  yaw_rate: nan
point_2: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  coordinate_frame: 0
  type_mask: 0
  position: 
    x: 0.0291219893843
    y: -0.0139899933711
    z: 0.00845336914062
  velocity: 
    x: nan
    y: nan
    z: nan
  acceleration_or_force: 
    x: nan
    y: nan
    z: nan
  yaw: -0.0730972737074
  yaw_rate: nan
point_3: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  coordinate_frame: 0
  type_mask: 0
  position: 
    x: 0.0291219893843
    y: -0.0139899933711
    z: 0.00845336914062
  velocity: 
    x: nan
    y: nan
    z: nan
  acceleration_or_force: 
    x: nan
    y: nan
    z: nan
  yaw: nan
  yaw_rate: nan
point_4: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  coordinate_frame: 0
  type_mask: 0
  position: 
    x: nan
    y: nan
    z: nan
  velocity: 
    x: nan
    y: nan
    z: nan
  acceleration_or_force: 
    x: nan
    y: nan
    z: nan
  yaw: nan
  yaw_rate: nan
point_5: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  coordinate_frame: 0
  type_mask: 0
  position: 
    x: nan
    y: nan
    z: nan
  velocity: 
    x: nan
    y: nan
    z: nan
  acceleration_or_force: 
    x: nan
    y: nan
    z: nan
  yaw: nan
  yaw_rate: nan
point_valid: [1, 1, 1, 0, 0]
command: [21, 21, 65535, 65535, 65535]
time_horizon: [0.0, 0.0, 0.0, 0.0, 0.0]
---
```

SwiftGust, 2020-01-21:
Landing with the planner while finding safe path not available since global planner is using altitude prior to assess risk of the cell https://github.com/PX4/avoidance/blob/master/global_planner/include/global_planner/global_planner.h#L39

And also would not allow altitude estimation drift & flying over declining surfaces which can be often found in real world