PX4/PX4-Avoidance #554: Avoidance node standalone

AhmedElsafy, 2020-01-23:
I am trying to catkin_make the avoidance_node to just rosrun it on its own without using the planners. I have seen in the CMakeList.txt file inside the Local_planner package this line 
`
## Declare a C++ executable
# add_executable(avoidance_node src/avoidance_node.cpp)
add_executable(local_planner_node src/nodes/local_planner_node_main.cpp)
`
I have uncommented this line but catkin_make is giving obviously the error of not finding the cpp file in src folder, is there away to do that ? or do I have to rewrite the CMakeList files myself for that to happen.

Thanks

Jaeyoung-Lim, 2020-01-23:
@AhmedElsafy Have you tried using `catkin build`? I believe `catkin_make` is near obsolete. catkin tools can be found [here](https://catkin-tools.readthedocs.io/en/latest/)

AhmedElsafy, 2020-01-23:
@Jaeyoung-Lim the error is exactly, 
<b>Cannot find source file</b>
as the catkin cant find where the source cpp files are, even with catkin build gives the same error.

Jaeyoung-Lim, 2020-01-23:
@AhmedElsafy Please show the whole log. You seem to have misconfigured the catkin_ws

AhmedElsafy, 2020-01-23:
@Jaeyoung-Lim I did actually change the CMakelist.txt in the local_planning package, as I was saying I un commented line 157 to build avoidance node so I can run the node on its own with out a planner.

Jaeyoung-Lim, 2020-01-23:
@AhmedElsafy Could you reference what line 157 means? It is possible to build just the avoidance node(without local and global planner) and use it with your planner. It was designed specifically for that usecase

AhmedElsafy, 2020-01-23:
@Jaeyoung-Lim line 157 is 
<b># add_executable(avoidance_node src/avoidance_node.cpp)</b>
if I leave that line as is the packages are built but I don't find any avoidance node to run as a standalone

Jaeyoung-Lim, 2020-01-23:
@AhmedElsafy If you remove that line, you remove the executable therefore you are not building the avoidance_node anymore

AhmedElsafy, 2020-01-23:
@Jaeyoung-Lim  right now that line is commented out in the CMakeList file on the repo, if I uncomment it the error appears and the node is not build !

https://github.com/PX4/avoidance/blob/ed63d10e41f0e985665d6578b9ae453995cf00c1/local_planner/CMakeLists.txt#L157