PX4/PX4-Avoidance #562: global_planner and px4

150202029, 2020-02-25:
Comments
@150202029 150202029 commented 2 days ago
Hello, I am having trouble running global_planner in the avoidance package. I would be happy if you help.

error

------------------------------------global_planner_sitl_mavros.launch--------------------------

<param name="use_sim_time" value="true" />

<node pkg="rqt_reconfigure" type="rqt_reconfigure" output="screen" name="rqt_reconfigure" />

<node pkg="tf" type="static_transform_publisher" name="tf_local_origin"
      args="0 0 0 $(arg mavros_transformation) 0 0 world local_origin 10"/>
<node pkg="tf" type="static_transform_publisher" name="tf_camera"
      args="0 0 0 -1.57 0 -1.57 fcu camera_link 10"/>

<include file="$(find px4)/launch/mavros_sitl.launch" >
    <arg name="world_path" value="$(arg world_path)" />
    <arg name="model" value="$(arg model)" />
</include>

<include file="$(find global_planner)/launch/global_planner_octomap.launch" >
    <arg name="use_three_point_msg" default="false"/>
    <arg name="point_cloud_topic" value="$(arg point_cloud_topic)" />
    <arg name="start_pos_x" value="$(arg start_pos_x)" />
    <arg name="start_pos_y" value="$(arg start_pos_y)" />
    <arg name="start_pos_z" value="$(arg start_pos_z)" />
</include>

<group if="$(arg use_three_point_msg)">
    <node name="bezier_mavros_wrapper" pkg="offboard" type="bezier_mapping.py" output="screen">
        <param name="bezier_duration" value="1.5" />
        <param name="cross_factor" value="2.0" />
        <remap from="/path/bezier_pt" to="/three_point_path" />
    </node>
</group>

<group if="$(arg use_stereo_simulation)">
    <node ns="stereo" pkg="stereo_image_proc" type="stereo_image_proc" name="stereo_image_proc"/>
</group>

<node pkg="rviz" type="rviz" output="screen" name="rviz" respawn="true"
      args="-d $(find global_planner)/resource/global_planner.rviz" />
----------------------------------mavros_sitl.launch----------------------------------------

<include file="$(find mavros)/launch/node.launch">
	<arg name="pluginlists_yaml" value="$(find mavros)/launch/px4_pluginlists.yaml" />
	<arg name="config_yaml" value="$(find mavros)/launch/px4_config.yaml" />

	<arg name="fcu_url" value="$(arg fcu_url)" />
	<arg name="gcs_url" value="$(arg gcs_url)" />
	<arg name="tgt_system" value="$(arg tgt_system)" />
	<arg name="tgt_component" value="$(arg tgt_component)" />
</include>

Hello, I am having trouble running global_planner in the avoidance package. I would be happy if you help.

[image]

150202029, 2020-02-25:
global_planner_sitl_mavros.launch--------------------------

<param name="use_sim_time" value="true" />

<node pkg="rqt_reconfigure" type="rqt_reconfigure" output="screen" name="rqt_reconfigure" />

<node pkg="tf" type="static_transform_publisher" name="tf_local_origin"
      args="0 0 0 $(arg mavros_transformation) 0 0 world local_origin 10"/>
<node pkg="tf" type="static_transform_publisher" name="tf_camera"
      args="0 0 0 -1.57 0 -1.57 fcu camera_link 10"/>

<include file="$(find px4)/launch/mavros_sitl.launch" >
    <arg name="world_path" value="$(arg world_path)" />
    <arg name="model" value="$(arg model)" />
</include>

<include file="$(find global_planner)/launch/global_planner_octomap.launch" >
    <arg name="use_three_point_msg" default="false"/>
    <arg name="point_cloud_topic" value="$(arg point_cloud_topic)" />
    <arg name="start_pos_x" value="$(arg start_pos_x)" />
    <arg name="start_pos_y" value="$(arg start_pos_y)" />
    <arg name="start_pos_z" value="$(arg start_pos_z)" />
</include>

<group if="$(arg use_three_point_msg)">
    <node name="bezier_mavros_wrapper" pkg="offboard" type="bezier_mapping.py" output="screen">
        <param name="bezier_duration" value="1.5" />
        <param name="cross_factor" value="2.0" />
        <remap from="/path/bezier_pt" to="/three_point_path" />
    </node>
</group>

<group if="$(arg use_stereo_simulation)">
    <node ns="stereo" pkg="stereo_image_proc" type="stereo_image_proc" name="stereo_image_proc"/>
</group>

<node pkg="rviz" type="rviz" output="screen" name="rviz" respawn="true"
      args="-d $(find global_planner)/resource/global_planner.rviz" />

150202029, 2020-02-25:
<param name="use_sim_time" value="true" />

<node pkg="rqt_reconfigure" type="rqt_reconfigure" output="screen" name="rqt_reconfigure" />

<node pkg="tf" type="static_transform_publisher" name="tf_local_origin"
      args="0 0 0 $(arg mavros_transformation) 0 0 world local_origin 10"/>
<node pkg="tf" type="static_transform_publisher" name="tf_camera"
      args="0 0 0 -1.57 0 -1.57 fcu camera_link 10"/>

<include file="$(find px4)/launch/mavros_sitl.launch" >
    <arg name="world_path" value="$(arg world_path)" />
    <arg name="model" value="$(arg model)" />
</include>

<include file="$(find global_planner)/launch/global_planner_octomap.launch" >
    <arg name="use_three_point_msg" default="false"/>
    <arg name="point_cloud_topic" value="$(arg point_cloud_topic)" />
    <arg name="start_pos_x" value="$(arg start_pos_x)" />
    <arg name="start_pos_y" value="$(arg start_pos_y)" />
    <arg name="start_pos_z" value="$(arg start_pos_z)" />
</include>

<group if="$(arg use_three_point_msg)">
    <node name="bezier_mavros_wrapper" pkg="offboard" type="bezier_mapping.py" output="screen">
        <param name="bezier_duration" value="1.5" />
        <param name="cross_factor" value="2.0" />
        <remap from="/path/bezier_pt" to="/three_point_path" />
    </node>
</group>

<group if="$(arg use_stereo_simulation)">
    <node ns="stereo" pkg="stereo_image_proc" type="stereo_image_proc" name="stereo_image_proc"/>
</group>

<node pkg="rviz" type="rviz" output="screen" name="rviz" respawn="true"
      args="-d $(find global_planner)/resource/global_planner.rviz" />

150202029, 2020-02-25:
------------------------------------global_planner_sitl_mavros.launch--------------------------

<param name="use_sim_time" value="true" />

<node pkg="rqt_reconfigure" type="rqt_reconfigure" output="screen" name="rqt_reconfigure" />

<node pkg="tf" type="static_transform_publisher" name="tf_local_origin"
      args="0 0 0 $(arg mavros_transformation) 0 0 world local_origin 10"/>
<node pkg="tf" type="static_transform_publisher" name="tf_camera"
      args="0 0 0 -1.57 0 -1.57 fcu camera_link 10"/>

<include file="$(find px4)/launch/mavros_sitl.launch" >
    <arg name="world_path" value="$(arg world_path)" />
    <arg name="model" value="$(arg model)" />
</include>

<include file="$(find global_planner)/launch/global_planner_octomap.launch" >
    <arg name="use_three_point_msg" default="false"/>
    <arg name="point_cloud_topic" value="$(arg point_cloud_topic)" />
    <arg name="start_pos_x" value="$(arg start_pos_x)" />
    <arg name="start_pos_y" value="$(arg start_pos_y)" />
    <arg name="start_pos_z" value="$(arg start_pos_z)" />
</include>

<group if="$(arg use_three_point_msg)">
    <node name="bezier_mavros_wrapper" pkg="offboard" type="bezier_mapping.py" output="screen">
        <param name="bezier_duration" value="1.5" />
        <param name="cross_factor" value="2.0" />
        <remap from="/path/bezier_pt" to="/three_point_path" />
    </node>
</group>

<group if="$(arg use_stereo_simulation)">
    <node ns="stereo" pkg="stereo_image_proc" type="stereo_image_proc" name="stereo_image_proc"/>
</group>

<node pkg="rviz" type="rviz" output="screen" name="rviz" respawn="true"
      args="-d $(find global_planner)/resource/global_planner.rviz" />

150202029, 2020-02-25:
<launch>
    <arg name="use_three_point_msg" default="false" />
    <arg name="use_stereo_simulation" default="false" />
    <arg name="mavros_transformation" default="0" />
    <arg name="world_path" default="$(find global_planner)/../sim/worlds/empty.world" />
    <arg name="point_cloud_topic" default="/camera/depth/points" />
    <arg name="start_pos_x" default="0.5" />
    <arg name="start_pos_y" default="0.5" />
    <arg name="start_pos_z" default="3.5" />
    <arg name="model" default="iris_depth_camera" />

    <param name="use_sim_time" value="true" />

    
    <node pkg="rqt_reconfigure" type="rqt_reconfigure" output="screen" name="rqt_reconfigure" />

    
    <node pkg="tf" type="static_transform_publisher" name="tf_local_origin"
          args="0 0 0 $(arg mavros_transformation) 0 0 world local_origin 10"/>
    <node pkg="tf" type="static_transform_publisher" name="tf_camera"
          args="0 0 0 -1.57 0 -1.57 fcu camera_link 10"/>

    
    <include file="$(find px4)/launch/mavros_sitl.launch" >
        <arg name="world_path" value="$(arg world_path)" />
        <arg name="model" value="$(arg model)" />
    </include>

    
    <include file="$(find global_planner)/launch/global_planner_octomap.launch" >
        <arg name="use_three_point_msg" default="false"/>
        <arg name="point_cloud_topic" value="$(arg point_cloud_topic)" />
        <arg name="start_pos_x" value="$(arg start_pos_x)" />
        <arg name="start_pos_y" value="$(arg start_pos_y)" />
        <arg name="start_pos_z" value="$(arg start_pos_z)" />
    </include>

    
    <group if="$(arg use_three_point_msg)">
        <node name="bezier_mavros_wrapper" pkg="offboard" type="bezier_mapping.py" output="screen">
            <param name="bezier_duration" value="1.5" />
            <param name="cross_factor" value="2.0" />
            <remap from="/path/bezier_pt" to="/three_point_path" />
        </node>
    </group>

    
    <group if="$(arg use_stereo_simulation)">
        <node ns="stereo" pkg="stereo_image_proc" type="stereo_image_proc" name="stereo_image_proc"/>
    </group>

    
    <node pkg="rviz" type="rviz" output="screen" name="rviz" respawn="true"
          args="-d $(find global_planner)/resource/global_planner.rviz" />
</launch>

mrivi, 2020-02-27:
Please follow these instructions https://github.com/PX4/avoidance#global-planner-advanced-not-flight-tested

closed 2020-02-27 by mrivi